Atmosphere-NX/Atmosphere #591: Updated to 8.1 using choidujour and now ams won't boot

melsbacksfriend, 2019-06-21:
Bug Report

What's the issue you encountered?
my switch boots normally after updating but I can't boot using atmosphère.  I just get a black screen
How can the issue be reproduced?
Update using choidujour and trying to boot
Crash Report
N/A
System Firmware Version
8.1.0
Environment?What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Fusée
Official release or unofficial build: Official release version 0.9.1
Do you have additional kips or sysmodules you're loading:
No
Additional context?Additional info about your environment: My switch is banned so I can't update normally.

shchmue, 2019-06-21:
Is your SD card formatted exfat? If so, did you install the exfat version of 8.1.0?

If not, what screens do you see before the black screen?

designgears, 2019-06-21:
Probably using the broken 0.9.1, download it again from releases page and give that a go.

melsbacksfriend, 2019-06-21:
I redownloaded it and it still isn't working.  It gets stuck after the third screen.

Specced, 2019-06-21:
What filesystem is your SDcard? Have you applied the Exfat-update?

jdcloz, 2019-06-21:
exactly the same problem here. after the "black ams logo" goes to black. i have a samsung 128gb, im in FAT32, and the update on 8.1 was whithout exfat ( normal). i was having problems whith choinx updating the emummc, by the way ( errors reading the some nca...maybe dont read it?)  but at least, in one of some tries... make it. but... you know the result.... or maybe that doesnt make the diference in that problem....

Specced, 2019-06-21:
Apply the Exfat update and try again.

jdcloz, 2019-06-21:
done in choinx ( i tried exfat too) but still fail whith the same error that i said early. when finally make the upgrade.... thats the result

nguyenbakhaihoan, 2019-06-22:
fixed this issue by using newest Atmosphere 0.9.1 but if you cannot launch any game if not overwrite newest sigpatches from https://gbatemp.net/attachments/2-0-0-8-1-0-7z.170513/ in atmosphere folder

fincs, 2019-06-22:
Links to piracy enabling files (i.e. sigpatches in this case) are not permitted.

jdcloz, 2019-06-22:
i was using the newest ams 0.9.1 and the problem going on

melsbacksfriend, 2019-06-22:
I can't build.  It gets stuck on Atmosphere-master/exosphere/src/configitem.c

leandroporto93, 2019-06-22:
Same problem here, already tried kosmos, injecting hekate.10.1 but wont work as well, with atmos 0.9.1 i get the black screen

leandroporto93, 2019-06-23:
Isn't the best solution, but I used the "boot.dat" from sxos, then used the choidujoir to downgrade the FW to 8.0.1, and solved the problem,

JeremyAgost, 2019-06-23:
For some reason I'm having the same issue. I originally had 5.1.0 installed. I'm preparing/copying to the SD card with a Mac, perhaps that's the issue?

jdcloz, 2019-06-23:
i have windows.

nguyenbakhaihoan, 2019-06-23:
As I mentioned above, my SD card was exFAT and I could boot atm using the newest version 0.9.1

JeremyAgost, 2019-06-23:
I reverted to my backup and then upgraded to 8.0.1 no problems. Is there any troubleshooting that we could do with 8.1.0 for you?

CypherNoodle, 2019-06-23:
"Step 2: Allow ChoiDujourNX to write to BIS"
https://switch.homebrew.guide/usingcfw/manualupgrade

JeremyAgost, 2019-06-23:
I don’t think that’s my issue. I had done that step. Plus, I was able to successfully update to 8.0.1 with the same SD configuration.

melsbacksfriend, 2019-06-24:
I switched to ReiNX and now I'm able to use cfw again.

melsbacksfriend, 2019-06-24:
Sorry, I didn't mean to do it twice.

closed 2019-06-24 by melsbacksfriend