Atmosphere-NX/Atmosphere #593: Error-Code 2000-0000 (0x0)

mrdude2478, 2019-06-22:
Switch firmware 6.2.0 (Atmosphere 0.9.1-master 3f7238cb).

Crashes after switch logo - if using via emunand ( on firmware 8.1.0), Atomosphere boots as normal.

So Atmosphere is no longer working on stock 6.2.0 firmware. Latest files from release downloads installed (along with bin file).

SciresM, 2019-06-22:
Please reopen using issue templates.

Also this is probably a user files error.

closed 2019-06-22 by SciresM

mrdude2478, 2019-06-23:
> Please reopen using issue templates.
> 
> Also this is probably a user files error.

Thanks, I couln't find the issue template, I did find that the crash was due to me using a custom theme for 8.1.0 (in emunand), and forgot to remove it when booting to stock 6.20. - So yes it was user error on my part. I do apologise for posting, kind regards & thanks for the update.

The3DPrintist, 2019-08-05:
this was my issue as well, thanks