Atmosphere-NX/Atmosphere #595: Atmosphere 0.9.1 crashes slightly after boot on FW 8.1.0

Kitsumi, 2019-06-23:
## Bug Report
### What's the issue you encountered?
Atmophere crashes about 30 seconds after booting up.

### Crash Report
- https://gist.github.com/Kitsumi/d0b95d0dc8a46748ba0e9bacb6bd3f9a

### System Firmware Version
- 8.1.0 Fat32

### Environment?
- Bootloader: Fusee chainloaded from Hekate 4.10.1 and Fusee using TegraRcmGUI
- Version: Atmosphere 0.9.1-master-3f7238cb
- Kips: default_nogc, fs_patches (Kosmos sigpatches)

### Additional context?
- 64GB SD Card formatted to fat32
- 5 burned fuses

I restored a NAND backup on FW version 4.1.0 and then used ChoiDujour to update to 8.1.0.

Kitsumi, 2019-06-23:
I just tried removing the sigpatches and it still crashes.

mrdude2478, 2019-06-23:
If you have a custom theme installed remove it - that's why mine was crashing - just rename titles folder first to test.

Kitsumi, 2019-06-23:
Actually, after looking over the crash report, it appears I did have more KIPs installed, specifically: ldn.mitm(I think?) I'm pretty sure this is Emuiibo

closed 2019-06-23 by Kitsumi

RenaKunisaki, 2019-07-10:
I'm having the same problem (I think? different error code) after updating to 8.1.0 (via ChoiDujourNX). It crashes after the boot logos instead of showing the home menu. If I remove all folders under `titles/0100*`, it works. I'm not sure what all of those are (various system titles, but no idea what the patches do).

Thing is, those aren't some extra app I installed; if I use Kosmos Updater to reinstall, it re-adds those folders, and the crash happens again. So (at least in my case) it seems to be a problem with Kosmos, not Emuiibo (which I don't have, unless it's part of Kosmos) or any other addon.

SciresM, 2019-07-10:
@RenaKunisaki kosmos includes a number of custom sysmodules, any one of which might have compatibility problems with 8.1.0.

RenaKunisaki, 2019-09-15:
For the sake of anyone else Googling: update your NxThemes.

Kitsumi, 2019-09-21:
And also update your sysmodules