Atmosphere-NX/Atmosphere #596: 0.9.1 reboots immediately

oslerw, 2019-06-23:
## Bug Report

### What's the issue you encountered?

Atmosphere 0.9.1 on FW 8.10 exfat gets most of the way to the boot, but shortly after the ams logo, I see the battery indicator and my console reboots into the stock firmware.

### How can the issue be reproduced?

1. Format SD card with exFAT (`mkfs.exfat` from `exfat-utils 1.3.0-1`)
2. Download the 0.9.1 release from the releases page (I just did this a few minutes ago, so it's up to date), uncompress it
3. Copy files over (`rsync -av`)
4. Place SD card in switch
5. RCM boot with Rekado (Android), fusee-primary.bin
6. sept logo
7. atmosphere logo
8. battery indicator in top right
9. black screen for 2 seconds
10. official switch logo
11. boots into official firmware

### Crash Report

Curiously, there are no crash reports on the SD card. I don't have a `/atmosphere/crash_reports` directory. I do have a `atmosphere/fatal_errors` directory, but it's entirely empty.

### System Firmware Version

8.1.0 (but I started with 8.0.1 which displayed the same behavior, I upgraded to exfat to see if it would help)</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: `fusee-primary.bin`
- Official release or unofficial build: 
  - Official release version 0.9.1
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: None

### Additional context?

- Additional info about your environment:
- Filesystem layout: https://gist.github.com/oslerw/b55bb55ca776713d4c597001a52b2977

oslerw, 2019-06-23:
Not sure if it's relevant, but I'm seeing the exact same behavior with ReiNX. I don't know what code might be shared between the two, they both appear to use sept at least.

Specced, 2019-06-23:
All current CFWs are based off Atmosphere, but there are small differences.

fincs, 2019-06-23:
The likes of ReiNX, Kosmos, SX OS; etc, are all basically rebranded versions of Atmosphère to various degrees. Usually they differ in the paths they use, or in security features disabled, or in piracy features added.

oslerw, 2019-06-23:
Ah, that makes sense. In that case I won't bother with the others since it'll probably be the same result.

Any idea how I can begin debugging this? Is there a way to get some logs or load a debug build or something?

oslerw, 2019-06-24:
This is really, really embarrassing, but as it turns out "rebooting into the stock firmware" was actually it successfully booting ams. I hadn't seen anyone using ams before, so I thought it was its own firmware with its own UI rather than patching the stock UI, so when it booted into ams I assumed something was wrong, when in reality that's just what it looks like.

Apologies for the bogus bug report, hopefully someone will find this when searching their issue and learn from my really dumb mistake.

closed 2019-06-24 by oslerw

SciresM, 2019-06-24:
@oslerw To be fair, Atmosphere doesn't "patch the stock UI" -- it doesn't really touch the UI at all.