Atmosphere-NX/Atmosphere #597: emuMMC not booting properly

McAlexica, 2019-06-24:
When I use the emummc.ini and change emummc_enabled to 1, it boots into sysnand, and it boots in emuNAND when I switch emummc_enabled to 0.

m4xw, 2019-06-24:
That.. makes no sense.
Can you share your emummc.ini ?

McAlexica, 2019-06-24:
Sure, do I do it here or in a new thread?

m4xw, 2019-06-24:
here

McAlexica, 2019-06-24:
5 minutes, please

McAlexica, 2019-06-24:
[emummc]
emummc_enabled = 1
emummc_sector = 0x371E800
emummc_id = 0x1

m4xw, 2019-06-24:
Are you sure, you don't just confuse sysNAND and emuNAND?

SciresM, 2019-06-24:
I agree you're probably just confusing the two.

If you delete the Nintendo_0001 folder, I would expect it to be recreated when booting with enabled = 1, and not recreated when booting with enabled = 0.

Do you observe something different?

McAlexica, 2019-06-24:
I mean, if it is set to 1 it should boot to emunand, right?
If it is set to 0, it should boot to sysnand

SciresM, 2019-06-24:
Yes. When set to 0 it doesn't inject emummc code into FS at all -- it's not possible for emunand to be booted with the setting at zero.

You're probably just confusing the two, hence why I suggested my test. What do you observe?

McAlexica, 2019-06-24:
I am right now

McAlexica, 2019-06-24:
Let's see

McAlexica, 2019-06-24:
No, I deleted the folder, the emummc is still set to 1 and when I boot, it goes into my old nand

McAlexica, 2019-06-24:
I followed the guides scratch by scratch, I don't think I messed up somewhere

SciresM, 2019-06-24:
"It goes into my old nand" -- why do you believe this is sysnand? Was the folder recreated?

SciresM, 2019-06-24:
It seems much more likely to me that you're booting into emunand and not realizing it's emunand.

McAlexica, 2019-06-24:
well I know because it is totally different from the nand I used for the past, 1 month

McAlexica, 2019-06-24:
I am probably just assuming, but, shouldn't emunand work on the files I already have? and sysnand work on the backup?
Otherwise I think it defeats the purpose :D

McAlexica, 2019-06-24:
Or maybe I am just really dumb and making people scratch their heads for no reason when they can do other more important stuff

SciresM, 2019-06-24:
I think it's probably literally just you thinking sysnand is emunand and vice versa.

You still haven't answered my question: was the folder recreated?

McAlexica, 2019-06-24:
So let me put this straight so I understand and maybe call this out, when emummc is enabled, I should be on cfw, I should be on my nand where I have my nsps, whatever, when I boot to sysnand, I should see my old games with no files, settings and such

McAlexica, 2019-06-24:
Yes, the folder was recreated, the Nitendo_0001 folder

SciresM, 2019-06-24:
1) "nsps" atmosphere does not support piracy/warez.

2) I think you fundamentally misunderstand how emummc works, at the time you create the partition it will be a copy of whatever you flashed to it. Divergences are then up to you

3) if the folder was recreated you're in emummc.

Closing because this is just user being wrong.

closed 2019-06-24 by SciresM