Atmosphere-NX/Atmosphere #599: Emunand via files doesn't work on FAT32 SD

shadow2560, 2019-06-25:
## Bug Report

### What's the issue you encountered?

Files Emunand doesn't launch on FAT32 SD but the same files work on EXFAT, with the same config file and environement.

### How can the issue be reproduced?

Create a folder on the SD (I use "emummc" folder), copy a splited dump on the folder "eMMC"  in the folder "emummc", rename the files like it should be (00, 01, 02, ..., boot0, boot1), create the correct "emummc.ini" file in "emummc" folder with the path to the emunand configured to "emmc" and the enabled value to "1" and try to launch the CFW via Fusee-primary.bin payload.

### Crash Report

N/A

### System Firmware Version

8.1.0

### Environment?

Atmosphere master build Self-compilation (Sept-secondary makefile configured correctly to compile with ".enc" files released in 0.9.1) launched by Fusee-primary payload.

### Additional context?

N/A

m4xw, 2019-06-25:
Invalid cluster size, must be 32k for FAT32.
Also not a Atmosphere Issue to begin with.

closed 2019-06-25 by fincs

shadow2560, 2019-06-25:
But my FAT32 SD is formated with 32K clusters size and it doesn't work and my other EXFAT SD work, the two are from the same manufactor and are exactly the same product.

m4xw, 2019-06-25:
I tested it with FAT32 myself.

shadow2560, 2019-06-25:
If this can help, my SD is a Samsung Evo +, maybe something is different from this card. I remember that when Linux launch started on Switch this SD has needed a specific kernel tweak to work properly, maybe something is related.

I'll make others tests like reformating the SD but my configuration seems to have no problem: config file is OK, archive byte is set, clusters size is good, files work on EXFAT then they are not corrupted or something like that and disk is not fragmented. I'll report back the results of my tests when they will be done.

m4xw, 2019-06-25:
> not a fake one, tested with H2test

This doesn't mean that it's not fake.
It just means that it can actually store as much data as was tested.

shadow2560, 2019-06-25:
Totaly exact but it's at least an info that the SD doesn't erase data "herself", it's a very important info  and from my remembering, read/write speed are also good.

shadow2560, 2019-07-01:
This problem was solved with the master build of the project (I haven't tested with 0.9.2 official release) without changing anything on my SD.

SciresM, 2019-07-01:
Guess this was more "using outdated FatFS" problems.