Atmosphere-NX/Atmosphere #601: EmuMMC can't boot: Error 2002-4364

ChickenNerd, 2019-06-25:
## Bug Report

### What's the issue you encountered?

So, I installed EmuMMC and atmosphere, used it a bit, enabled incognito, installed an nsp file and tried to run it. I got a crash and rebooted. Everything was fine and when i tried to reboot it no longer boots. (Error 2002-4364)

### How can the issue be reproduced?

1. Install atmosphere, EmuMMC and ams_mitm.
2. Enable Incognito,
3. Install an NSP with Goldleaf,
4. Try to run the NSP.
5. It crashes after x(1-10) run.
6. Restart console.
7. After Switch logo the error report appears.

### Crash Report

https://gist.github.com/ChickenNerd/c30e8b82af229e3887536b9434061dc5

### System Firmware Version

8.1.0

### Environment?

- What bootloader: fusee-primaty.bin
- Official release or unofficial build: Official 0.9.1
- Do you have additional kips or sysmodules you're loading: ams_mitm.
- Homebrew software installed: 
Goldleaf, Tinfoil, ftpd (not system module), hbappstore
, mgba, retroarch, ChoiDjourNX, EdiZon, Checkpoint, NXThemes-installer, emuMMC-Toggle

### Additional context?

N/A

fincs, 2019-06-25:
Piracy is not supported.

closed 2019-06-25 by fincs

ChickenNerd, 2019-06-25:
ummm, it was an forwarder

ZachyCatGames, 2019-06-25:
> ummm, it was an forwarder

Forwarders aren't supported either

ChickenNerd, 2019-06-25:
ok then

fincs, 2019-06-25:
Homebrew is launched using the homebrew menu, not through dodgy "forwarder" titles.

GRAnimated, 2019-06-25:
I know NSPs are a guaranteed ban but why are forwarders dodgy? They've been shown to work fine,(even if they aren't as practical as HBL) and there's no point in making a brick forwarder (and if someone did, you can easily check).

fincs, 2019-06-25:
Forwarders are no different to other NSPs and carry the same risks. Please stick to the officially supported way of running homebrew and spare everyone the trouble.

pixel-stuck, 2019-06-25:
@GRAnimated 
they: 
 - require patches used to enable piracy
 - may be just using "forwarders" as an excuse
 - carry similar ban risks
 - are not officially supported with AMS (requires the aforementioned piracy patches)

If you want support, and can show this is an issue without external AMS patches, we'd be happy to provide support.