Atmosphere-NX/Atmosphere #602: Add 'On/Off Switch' codetype (CC) from Wii's ocarina codehandler to AMS's cheat engine

M1CR0H4CK3R, 2019-06-26:
## Feature Request

### What feature are you suggesting?
#### Overview:
- I am requesting that  support be added for the 'On/Off Switch' codetype from the Wii's ocarina codehandler. The codetype is detailed [here](https://geckocodes.org/index.php?arsenal=1#CC) at geckocodes.org.</br>

#### Smaller Details: 
- When the code's line is CC000000 00000001, the switch is on by default, and when the line reads CC000000 00000000, it is off by default. A use case is for when you want to use a button combo to enable a code, and use the same button combo to disable it, whether the values to write are at the same addresses, or at different addresses altogether from one another. The link I put in the section above explains how to implement it pretty well by specifically saying how it works</br>

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
I'm not sure if making codes should be considered as 'for the end-user' or not, so I'll just put it here. Heh. :
It makes AMS's cheat engine more versatile, and therefore extends its useability.

</br>

SciresM, 2019-06-26:
This would be...difficult -- no state is preserved whatsoever between instances of the VM. Adding that kind of nonvolatile storage would be a major architectural change...

SciresM, 2019-06-26:
I read over the description for the code type on the gecko site -- I do not understand what it does or how it works. 

What exactly does this do?

piepie62, 2019-06-26:
I've been looking at it, too, with Luma's cheat engine in mind. My interpretation is that when the internal toggle is false, any opcode that would do any actual state edit doesn't actually get executed, instead just being skipped. All conditionals would still be executed on every pass the cheat is "activated", making the pseudocode for this something like this:
`void cheatCode() {
    static bool isOn = true;
    if (codetype == 0xCC)
        isOn = !isOn;
    if (codetype == editCodetype && isOn)
        doEdit();
    if (codetype == conditional)
        doConditional();
}`

SciresM, 2019-06-26:
Ahh....that is a *bizarre* design. That also does not play well with the current cheat VM, which does not separate or provide guarantees about the execution order of individual cheats.

piepie62, 2019-06-26:
By execution order of individual cheats, do you mean individual opcodes? Because my interpretation would be within a cheat; that pseudofunction was supposed to represent an independent cheat code

piepie62, 2019-06-26:
Assuming that an entire cheat's (defined as a single named group of opcodes) opcodes are executed in sequence, it seems like a sane design to me

SciresM, 2019-06-26:
No, I mean cheats. The architecture currently concatenates all cheats and executes then as a single program, guaranteeing the master code runs first and providing no other guarantees. This is necessary to keep backwards compat with gateway's spec and existing codes.

SciresM, 2019-06-26:
There is no concept of cheat isolation/boundary, and thus a cheat specific switch isn't really implementable...

You could do it VM wise but then that's weird and the lack of execution ordering guarantees will get fucky.

RenaKunisaki, 2019-11-02:
I'm pretty sure that's how Gecko works too. Every line of a code is executed in order, but all codes are concatenated. If one code leaves things in a bad state, others won't work. This is actually sometimes desirable, as you can toggle/alter the behaviour of other cheats, assuming they're concatenated in a known order.

SciresM, 2019-12-31:
I still don't understand what this code type does/how it works. If someone explains it to me on discord and gives a concrete use case for it I'm not averse to adding it.

SciresM, 2019-12-31:
Closing for now.

closed 2019-12-31 by SciresM