Atmosphere-NX/Atmosphere #605: HACK NOT WORKING

BZA-B, 2019-06-29:
We would like to use GitHub to keep track of problems/feature requests.  
If you have a question, please join the ReSwitched discord for help. 
- Discord link: https://discordapp.com/invite/DThbZ7z 

Hey m8,

I msgd u on Reddit too but not sure if u got it. I basically updated my Switch to latest firmware after seeing its been hacked. I was on 7.0.0 which was working great. So I inject the payload then after the Atmosphere logo I go into Album and its just the stock Album :( I'm not sure what I'm doing wrong. I've used a different SD and added fresh files but does the same thing. Any help is massively appreciated. I havn't seen anyone else with this issue.

Kind regards,
BZA79

Bappsack, 2019-06-29:
Hold R while launching Album

SciresM, 2019-06-29:
1) Use Issue Templates
2) Hold R.

closed 2019-06-29 by SciresM

BZA-B, 2019-06-29:
If this works I'll literally jump to the ceiling :) Cheers for your response...here goes nothing!

BZA-B, 2019-06-29:
Ok, that kinda worked! :) well its loaded hbmenu but only 'reboot to payload' app is there? Not sure what to do?

BZA-B, 2019-06-29:
Yeh so none of my apps are in there :( I clicked on it. It reloads and Atmosphere loads with SEPT across the screen then I just gotta do the same thing. Umm, am I doing things right? How do I get homebrew store n Gold leaf n everything (apps basically) in hbmenu? Do I have to manually add them to my SD? I just expected to see everything I suppose. There is a Netloader option I've noticed? I was scared to try that. Anything I don't know, I don't tingle with.

BZA-B, 2019-06-29:
Ok I'm using my other SD now which has all my stuff on it. Sorry I was using the fresh SD card I made with nothing on it but the new Atmosphere.

BZA-B, 2019-06-29:
Holy jesus :D