Atmosphere-NX/Atmosphere #608: [FW4.1.0] Can't get past Nintendo switch logo on 0.9.2

fOmey, 2019-06-30:
### What's the issue you encountered?

Can't get past the Nintendo switch logo while using 0.9.2.. 0.9.1 seems to boot fine.

### System Firmware Version

4.1.0

### Environment?

Default stock atmosphere setup

### Misc

I have tried formatting microsd, both exfat & fat32, re-transferring all the files to no avail.. as soon as I tried 0.9.1 it booted up fine.

jejer, 2019-06-30:
black screen after sept logo, seems 0.9.1 also failed to boot. 
I did a nand restore from 5.1 to 4.1.

jermafenser, 2019-06-30:
I have a similar issue as fOMey. My 4.0.1 system hangs after the switch logo. I'm using Kosmos 13 and CTCaer hekate 5.0.0. Booting into "stock" from hekate works as expected.

Raugo, 2019-06-30:
Same here, the switch get stuck in the nintendo switch logo. The mmc is in 4.1 and the emummc is on 8.0.1. With 0.9.1 works well.

jul2003, 2019-06-30:
Same for me on 4.1.0 atmosphere release started with fusee-primary.bin

hexkyz, 2019-06-30:
Based on these descriptions it appears to be a regression introduced somewhere by the stratosphere re-write.
Just to be clear, does atmosphere 0.9.2 boot at all in 4.x *without* emuMMC?

jul2003, 2019-06-30:
I don’t use emuMMC

gerritb, 2019-06-30:
Same here. OFW without emuMMC works fine, hangs at Nintendo logo on emuMMC (both CFW and Stock)

xwinno, 2019-06-30:
Happening here too, stuck at Nintendo logo on emuMMC.

Le0T, 2019-06-30:
Same thing here.
Can't get past the Nintendo Switch boot logo on emuMMC partition based, everything created by hekate.
HOS Version: 4.0.1

Booting Atmosphère just with this hekate_ipl.ini entry:

fss0=atmosphere/fusee-secondary.bin

jetracer, 2019-06-30:
Tried atmohsphere, cant load past nintendo icon. Reinx, loads boot logo shows nintendo and restarts back to ofw, sxos boots and yellow screens once in hbmenu.

ProtoxiDe22, 2019-06-30:
Same problem here, using fusee and hekate 5.0.0, ofw 4.0.1, created the emummc from hekate, when i try booting cfw on either the emummc or the sysnand, the console hangs on the switch logo

Strat00s, 2019-06-30:
Same here. But I am on 1.0.0. It just hangs on Nintendo logo, as long as I am using emummc (file based). Booting without EmuMMC works fine.

Lordmau5, 2019-06-30:
I'm on FW 4.0.1 and Atmosphere 0.9.2 and can confirm that I am stuck on the Nintendo Switch logo as well.

biom4st3r, 2019-06-30:
iPatch 4.1:
 0.9.2: 
     - emuMMC CFW: Stuck at Nintendo Boot logo
     - non-emu CFW: Same
     - emuMMC stock: black screen
     - non-emu stock : black Screen

PabloGBarcelo, 2019-06-30:
Same error (stuck logo Nintendo Switch) in FW 4.1.0 And Atmosphere 0.9.2

drilus, 2019-07-01:
Same exact issue for me using Atmosphere 0.9.2 with FW 4.1.0.

AMS 0.9.1 boots just fine with 4.1.0.

Bumblecito, 2019-07-01:
Same here.
Non-patched
OFW: 4.1.0
Atmosphère 0.9.2
Hekate 5.0.0

Tried both emuMMC (SD Files & Raw)

Atmosphère/Kosmos emu and sys: stuck on Nintendo Switch logo
Stock sys: work
Stock emu: black screen
Other CFW (SX OS): work

rocket0634, 2019-07-01:
Would like to also report the same issue in Atmosphere 0.9.2 with FW 4.1.0 on emummc.

0.9.1 and Hekate 5.0.0 can boot it just fine.

InstinctXIV, 2019-07-01:
Same Bought mine yesterday.

Non-patched
OFW: 4.1.0
Atmosphere 0.9.2
Hekate 5.0.0

Reformatted to EXFAT and FAT32 no changes, stuck on Nintendo logo.

aranwarez, 2019-07-01:
4.1.0 get stuck on Nintendo Switch logo. Atmosphere 0.9.1 working fine.. No emuMMC used..

ckurtz22, 2019-07-01:
Same issue, 4.1.0, 0.9.2, freezes on Switch logo. Can try reverting certain sysmodules until it works

ckurtz22, 2019-07-01:
Alright, I've confirmed through replacement of every other kip individually, the issue lies with spl

ckurtz22, 2019-07-01:
https://github.com/Atmosphere-NX/Atmosphere/pull/616 Fix here in this PR.

SciresM, 2019-07-01:
Closed thanks to @ckurtz22

closed 2019-07-01 by SciresM