Atmosphere-NX/Atmosphere #609: Crash when try to open the HBL via album while an application is running

Maschell, 2019-06-30:
## Bug Report
### What's the issue you encountered?
When I try to open the HBL via the album while an application is running, I get an Error-Code (2345-0032 (0x4159). Sometimes the console even crashes completly and I have to force a shutdown. Using the "atmosphere-0.9.2-master-b0a66a63+hbl-2.1+hbmenu-3.0.1.zip" release package.
[image]

It does work fine on "atmosphere-0.9.1-master-241b8f46+hbl-2.1+hbmenu-3.0.1.zip"

### How can the issue be reproduced?
Start any application and try to open the HBL via the album.

Here is a video: https://streamable.com/30rsn
rebooting from a crash, opens the HBL fine, start and close application, opens hbl fine, start application, open hbl does crash

### Crash Report
It only procudes an fatal_report:
```
Atmosphère Fatal Report (v1.0):
Result:                          0x4159 (2345-0032)

Title ID:                        010000000000100d
Firmware:                        8.0.1 (Atmosphère 0.9.2-master-0d840e19)
General Purpose Registers:
    PC:                          0000000000000000
Start Address:                   0000000000000000
Stack Trace:
```

### System Firmware Version
8.0.1

### Environment?
Latest fusee-primary.bin

SciresM, 2019-06-30:
It seems...unlikely that Atmosphere is causing HID initialization issues in HBMenu? I'll talk about it with switchdev team.

SciresM, 2019-06-30:
Can you confirm issue occurs when setup is clean atmosphere without anything from unofficial websites? (Without sdsetup)

NickHam13, 2019-07-01:
I tried this and everything worked as intended. You should try grabbing the releases from only verified sources (as stated above).

SirAIDS, 2019-07-01:
This happens to me too. Everything is taken directly from github so should be verified. I didn't use SDsetup

SirAIDS, 2019-07-01:
Just on a hunch I tried reinstalling hekate 5.0.0 and I am still getting the same error. However my error code is slightly different. It is Error Code: 2345-0031 (0x3f59). Everything else matches the screenshot that the original reporter has posted

SirAIDS, 2019-07-01:
I just tried downgrading to 0.9.1 with the package from github and got a different fatal error under the same conditions, still using the current hekate. I am no coder but this one seems more informative so I'll attach the fatal error file:

[01561959260_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/3344058/01561959260_010000000000100d.log)

Maschell, 2019-07-01:
I just a clean sd card, with the atmosphere-0.9.1-master-241b8f46+hbl-2.1+hbmenu-3.0.1.zip, works fine.
Afterwards I made some more tests:
- works after completly deleting the atmosphere folder and copy it from the zip
- works after just deleting the atmosphere/titles folder and copy it from the zip

it looks like I had something in the titles folder that 0.9.2 didn't like. I once added cheats and did run the Edizon updater. The files directly coming from atmosphere were replaced with the one from the zip (and have still the same hash).

While updating atmosphere to 0.9.2 I also (tried to) setup emuMMC and added a new partition to the sd card. This may have corrupted some files?  It's still weird that 0.9.1 worked fine. 

However: a completly clean atmosphere/titles folder fixed it for me. I added my old titles folder in case somebody wants to reproduce it / check what files caused it
[old_titles.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3344779/old_titles.zip)

SciresM, 2019-07-01:
Yeah, every time this gets reported it turns out to be SD issues.

closed 2019-07-01 by SciresM