Atmosphere-NX/Atmosphere #610: emuMMC not working with exFAT

Niktendo, 2019-06-30:
I started with shrinking my SD Card (formatted with exFAT), installed emuMMC  and everything works perfectly. But when I choosed either CFW emuMMC (Stock doesn't work at all as it requires sept, secmon and a warmboot binary at 8.1.0) or CFW/Stock with disabled emuMMC, atmosphere stucks at the "sept" splash screen. I disabled emuMMC in Hekate and tried to boot the fusee-primary payload and it stucks at "Fatal error: Failed to stat stage2 (atmosphere/fusee-secondary.bin). It seems like this is the text-based output of the splash screen I saw before, because when I press the power button, it also ends up in a fatal-error/splash screen-loop. It also seems like atmosphere doesn't recognizes my SD card but Hakate says that emuMMC is installed. 
I'm currently running stock 7.0.0 (9 burnt fuses) with the exFAT Update and emuMMC (SD Raw Partition) at version 8.1.0. I tried the whole thing with a FAT32 formatted SD Card et voilà, it worked perfectly.

hexkyz, 2019-06-30:
Did you have the atmosphere/fusee-secondary.bin file in the SD card when that error ("Failed to stat stage2") popped up?

Niktendo, 2019-06-30:
Yes, of course. I also replaced it with the newest one from atmosphere 0.9.2 (the same as the Kosmos one).

ubenmackin, 2019-07-01:
I'm having the same issue. My card was formatted for exFat and was working fine. I did a shrink and created a new partition. Hekate loads up, and I was able to create the emummc partition fine. I can run the archive bit repair as well, and can see it iterating through all the files on the SD card.

If I try and boot with either emummc or cfw without emummc, I get a hang at the sept logo. Booting with fuse-primary, shows the same text (can't stat stage2). I've double checked, and the file is there. I even tried deleting and replacing the atmosphere folder, but same thing happens.

If I just do a stock boot, it boots up fine, and I can review the SD card via DataManagement (and it shows my games, and free space).

Let me know if there is anything else I can pull down.

hexkyz, 2019-07-01:
It's suspected that this issue may be due to fusee and sept using an outdated version of FatFs (which had an exFAT related bug). FatFs has been updated in the latest commits, so hopefully this should be fixed in the next release.

ubenmackin, 2019-07-01:
Awesome @hexkyz. Looking forward to trying it out once it is released.

SciresM, 2019-07-01:
Closing because fatfs stuff should have this resolved. If it's not, will reopen.

closed 2019-07-01 by SciresM