Atmosphere-NX/Atmosphere #612: Games & applications does not launch in loose-file based emuMMC environment

FusionFernia0000, 2019-06-30:
## Bug Report

### What's the issue you encountered?

Games & applications does not launch in loose-file based emuMMC environment.

### How can the issue be reproduced?

1. Create a folder named "stock_clean" in /sdcard/emuMMC/
2. Move rawnand.bin (30-ishGB), BOOT0(4MB) and BOOT1(4MB) to /sdcard/emuMMC/stock_clean/eMMC
3. Rename rawnand.bin to 00
4. Move the Nintendo folder (from the same backup) to /sdcard/emuMMC/stock_clean
5. Create a file named file_based in /sdcard/emuMMC/stock_clean (for Hekate to register)
6. On Switch, boot Hekate v5.0.0 from RCM
7. In Hekate emuMMC management, switch active emuMMC to stock_clean
8. Chainload fusee-primary.bin from Hetake
9. Wait for HorizonOS to boot (File-based emuMMC is rather slow)

Once booted, attempt to launch a game will result in the message "Save data is corrupted, scan for corrupted data?", after scanning and deleting the save data, another message saying "Game data is corrupted, scan for corrupted data?"

If using another user account that doesn't have a save on the application, the application crashed immediately, selected "Do not send data to Nintendo", launch the app again, app still crashed, but no option to report to Nintendo.

Game cartridges seem to boot fine, however.

Tried another EMMC backup (redone the steps above), still has the same problem.

Games and apps launched fine in sysMMC and an emuMMC created using Hekate (a carbon copy of sysMMC, loose-file based)

All installed games and apps are obtained legally through official methods, no piracy or "backups" going on here.

### Crash Report

Atmosphere didn't crash.

### System Firmware Version

emuMMC with aforementioned problem : 5.1.0
sysMMC : 7.0.0
emuMMC "SD00" (Carbon copy of sysMMC created using Hekate) : 7.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: 
Hekate v5.0.0, then chainload fusee-primary.bin

- Official release or unofficial build: 
0.9.2, official

- Do you have additional kips or sysmodules you're loading:
sys-clk installed

- Homebrew software installed: 
Checkpoint, Retroarch, NXShell

### Additional context?

- Additional info about your environment:
I noticed emuMMC created by Hekate has the rawnand.bin split into files named "00" to "07", while the one i moved in only has a "00" around 30GB in size. I don't think that would be the cause of the problem though.

FusionFernia0000, 2019-07-01:
Did some more testing, I first restored my EMMC backup to sysMMC, all apps launched as expected. I then created a emuMMC with Hekate, the newly created emuMMC also worked fine.

SciresM, 2019-07-01:
```
Did some more testing, I first restored my EMMC backup to sysMMC, all apps launched as expected. I then created a emuMMC with Hekate, the newly created emuMMC also worked fine.

```

I'm just gonna guess there's some savedata mismatch in the other backup, and so the console doesn't think it's entitled to play whatever games. Maybe mis-redirection of Nintendo dir. This is *especially* true since you said game cartridges boot fine.

Either way, sounds like there's no actual problem.

closed 2019-07-01 by SciresM