Atmosphere-NX/Atmosphere #615: Cheat not working

YuliaS11, 2019-07-01:
After updating to Atomosphere 0.9.2, the cheat function seems to be broken and a previously working cheat txt is no longer recognized, not executed or read by Edizon (goldleaf suggest both the directory and the file are valid). It is however working for 0.9.1.

xFalox, 2019-07-01:
After the 0.9.2 update, Edizon can't read the BuildId from memory anymore, instead it reads a random memory location which constantly changes.
Most likely a changed api is causing this and Edizon just needs to be updated.

WerWolv, 2019-07-01:
That's been fixed in one of the last commits. Compile Atmosphere yourself or wait till the next release for the cheats to work again

closed 2019-07-01 by SciresM