Atmosphere-NX/Atmosphere #619: build error on Atmosphere

Kronos2308, 2019-07-01:
# Bug Report
I have a problem trying to compile Atmosphere
the error looks like this

GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c: In function 'vsnprintf':
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c:885:19: error: array subscript has type 'char' [-Werror=char-subscripts]
  885 |    while (isalnum(*fmt))
      |                   ^~~~
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c: In function 'vsscanf':
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c:1459:15: error: array subscript has type 'char' [-Werror=char-subscripts]
 1459 |   if (isspace(*fmt)) {
      |               ^~~~
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c:1479:20: error: array subscript has type 'char' [-Werror=char-subscripts]
 1479 |    while (!isspace(*fmt) && *fmt != '%' && *fmt)
      |                    ^~~~
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c:1481:20: error: array subscript has type 'char' [-Werror=char-subscripts]
 1481 |    while (!isspace(*str) && *str)
      |                    ^~~~
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c:1534:28: error: array subscript has type 'char' [-Werror=char-subscripts]
 1534 |    while (*str && !isspace(*str) && field_width--)

I do not know why it happens but I can not compile it
I have asked and nobody knows what happens
I have the last commit of Atmosphere, the last libnx
and tried to erase and clone atmosphere again
but nothing, I would greatly appreciate a little help

hexkyz, 2019-07-01:
This should now be fixed on the latest commit.

Kronos2308, 2019-07-01:
ok

Kronos2308, 2019-07-01:
@hexkyz nop

vsnprintf':
GitHub/Atmosphere/sept/sept-secondary/src/lib/vsprintf.c:885:19: error: array subscript has type 'char' [-Werror=char-subscripts]
  885 |    while (isalnum(*fmt))
      |                   ^~~~
cc1.exe: all warnings being treated as errors
make[3]: *** [/opt/devkitpro/devkitARM/base_rules:85: vsprintf.o] Error 1
make[2]: *** [Makefile:129: build] Error 2
make[2]: Leaving directory '/home/Administrador.DESKTOP-4QS9TJ8/Documents/GitHub/Atmosphere/sept/sept-secondary'
make[1]: *** [Makefile:8: sept-secondary] Error 2
make[1]: Leaving directory '/home/Administrador.DESKTOP-4QS9TJ8/Documents/GitHub/Atmosphere/sept'

now i got this error

SciresM, 2019-07-01:
@hexkyz we should probably update to r53 ourselves.

SciresM, 2019-07-01:
Gonna close this. I'll update my PC to r53 tonight and clear up any issues then.

closed 2019-07-01 by SciresM

Kronos2308, 2019-07-01:
tanks

hexkyz, 2019-07-02:
Latest commit is confirmed to be building with devkitARM r53.