Atmosphere-NX/Atmosphere #620: Switch crashes when trying to open the Quick Home Menu

Pyjamaa, 2019-07-02:
## Bug Report
Switch crashes when trying to open the quick Home Menu

### What's the issue you encountered?
Why I try to open the quick home menu whenever I'm in a game or in the home menu then the Switch crashes. Since I updated from FW 6.2 to FW 8.1.0

### How can the issue be reproduced?

Hold the Homebutton anywhere to open the quick home menu.

### Crash Report
It didn't create a normal crash report only a fatal report

https://gist.github.com/Pyjamaa/6a04900265681afbdfecc806f312dbea

### System Firmware Version

FW 8.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: offical
  - Atmosphère 0.9.1
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment:
Was fine when I was on FW 6.2 and Atmos 0.8.3
Only started to happen when I updated to 8.1.0 and atmos 0.9.1 like a week ago

SciresM, 2019-07-02:
Does error occur on 0.9.2? Would appreciate fatal report from latest release build.

Pyjamaa, 2019-07-02:
Let me update real quick.
Then I'm gonna post the new log

Pyjamaa, 2019-07-02:
Still the same on 0.9.2
Here is the crash report

https://gist.github.com/Pyjamaa/52fd359915b50a70f2972d69c2db66e3

SciresM, 2019-07-02:
Just to check: you don't have anything non-standard in /atmosphere/titles, correct? No custom themes or whatever?

Pyjamaa, 2019-07-02:
I have a custom theme installed.
So I guess I should delete everything from the title folder besides the standard folders and retry?

SciresM, 2019-07-02:
Yes. I am going to guess this is a custom theme problem. The error report has FLAN magic in its registers, which indicates to me something is going wrong in parsing layout, lol. Test?

Pyjamaa, 2019-07-02:
Ok after recreating the title folder it works now.
Guess that was from the time when I was experimenting with leagcy themes which went all in the title folder. :sweat_smile:
My bad :sweat_smile:

Thanks for the help ScriesM!

closed 2019-07-02 by Pyjamaa

SciresM, 2019-07-02:
No problem, glad it's resolved ^^