Atmosphere-NX/Atmosphere #622: Atmosphere Error 2000-000 on Launch

lung9, 2019-07-06:
### **What's the issue you encountered?**

Starting up Atmosphere only shows Error Message after Nintendo Logo. (Error 2000-0000)
[image]

### **How can the issue be reproduced?**

To be honest Im not sure. It started after i downgraded to 6.2.0

### **Crash report?**
[01562431238_010000000000003e.log](https://github.com/Atmosphere-NX/Atmosphere/files/3364862/01562431238_010000000000003e.log)
### 
**Environment?**

- What bootloader (fusee, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: Atmosphere 0.9.2
- Do you have additional kips you're loading: no
- Additional info about your environment: firmware version 6.2.0

SciresM, 2019-07-06:
You downgraded in a way that caused the olsc system module to have invalid savedata.

Delete the relevant savedata or restore a backed up copy from before you updated -- this is a requirement for downgrading.

closed 2019-07-06 by SciresM

lung9, 2019-07-06:
Where can i find these savefiles?