Atmosphere-NX/Atmosphere #624: unable to build after updating with dkp-pacman

urherenow, 2019-07-07:
## Bug Report

### What's the issue you encountered?

Building fails with loader. Specifically with ldr_process_creation.cpp

### How can the issue be reproduced?
1) run "sudo dkp-pacman -Syuu"
2) Pull latest switchbrew/libnx and run "make" then "sudo -E make install"
3) switch to Atmoshpere folder and "make"

### Crash Report
N/A

### System Firmware Version

N/A

### Environment?

Self-compilation (with most ams commits successfully compiled up to around mid-may)
     - Ubuntu VM
     - devkitARM r53-1
     - devkitA64 r13-4

### Additional context?

on Discord it was suggested to roll back one commit on libnx, so I tried that as well. 61c0f4b

The specific errors I get are "not declared in this scope":

/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:201:30: error: 'TitleId_Settings' was not declared in this scope
                 (title_id == TitleId_Settings ||
                              ^~~~~~~~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:201:30: note: suggested alternative: 'Module_Settings'
                 (title_id == TitleId_Settings ||
                              ^~~~~~~~~~~~~~~~
                              Module_Settings
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:202:30: error: 'TitleId_Bus' was not declared in this scope
                  title_id == TitleId_Bus ||
                              ^~~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:203:30: error: 'TitleId_Audio' was not declared in this scope
                  title_id == TitleId_Audio ||
                              ^~~~~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:204:30: error: 'TitleId_NvServices' was not declared in this scope
                  title_id == TitleId_NvServices ||
                              ^~~~~~~~~~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:204:30: note: suggested alternative: 'MitmQueryService'
                  title_id == TitleId_NvServices ||
                              ^~~~~~~~~~~~~~~~~~
                              MitmQueryService
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:205:30: error: 'TitleId_Ns' was not declared in this scope
                  title_id == TitleId_Ns ||
                              ^~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:205:30: note: suggested alternative: 'title_id'
                  title_id == TitleId_Ns ||
                              ^~~~~~~~~~
                              title_id
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:206:30: error: 'TitleId_Ssl' was not declared in this scope
                  title_id == TitleId_Ssl ||
                              ^~~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:207:30: error: 'TitleId_Es' was not declared in this scope
                  title_id == TitleId_Es ||
                              ^~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:207:30: note: suggested alternative: 'title_id'
                  title_id == TitleId_Es ||
                              ^~~~~~~~~~
                              title_id
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:208:30: error: 'TitleId_Creport' was not declared in this scope
                  title_id == TitleId_Creport ||
                              ^~~~~~~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:208:30: note: suggested alternative: 'Module_Creport'
                  title_id == TitleId_Creport ||
                              ^~~~~~~~~~~~~~~
                              Module_Creport
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:209:30: error: 'TitleId_Ro' was not declared in this scope
                  title_id == TitleId_Ro);
                              ^~~~~~~~~~
/home/urherenow/projects/Atmosphere/stratosphere/loader/source/ldr_process_creation.cpp:209:30: note: suggested alternative: 'title_id'
                  title_id == TitleId_Ro);
                              ^~~~~~~~~~
                              title_id
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:14: ldr_process_creation.o] Error 1
make[2]: *** [Makefile:125: build] Error 2
make[2]: Leaving directory '/home/urherenow/projects/Atmosphere/stratosphere/loader'
make[1]: *** [Makefile:10: loader] Error 2
make[1]: Leaving directory '/home/urherenow/projects/Atmosphere/stratosphere'
make: *** [Makefile:21: stratosphere] Error 2

fincs, 2019-07-07:
Not a bug. Atmosphère presently depends on an unreleased version of libnx. Please be patient until we release the next version of libnx.

closed 2019-07-07 by fincs

misson20000, 2019-07-07:
@fincs he's building with libnx from git

urherenow, 2019-07-07:
Correct. Both the lattest commit from master, and rolled back one commit as suggested in discord.

fincs, 2019-07-07:
Looks like there's somehow a mismatch between the libstrat source code and the Atmosphère source code.

In general, it isn't really recommended for the general public to build development snapshots of projects precisely because issues like these can occur with multiple moving parts. Right now there's a different branch in development and all focus is currently on it. In addition, building and installing libnx from git is a practice we discourage because it can cause random issues, make build environments hard to reproduce and in addition confuses the package manager, placing the toolchain installation into an unpredictable state.