Atmosphere-NX/Atmosphere #627: Atmosphère crash after activating emuiibo and other sys modules

j-james, 2019-07-08:
## Bug Report

### What's the issue you encountered?

Atmosphère crashed.

### How can the issue be reproduced?

It can be reproduced by going to sdsetup.com](https://www.sdsetup.com/console?switch) and selecting the same packages, then turning on emuiibo, sys-ftpd, and sys-clk and rebooting into CFW.

### Crash Report

I have two crash reports, one after booting from fusee-primary and the other after booting from hekate. [Gist](https://gist.github.com/j-james/ec1a7b6b13f9ad1c207557bd8d5e76b7)

### System Firmware Version

8.1.0

### Environment?

I launched Atmosphère by fusèe first, then hekate.\
I used Kosmos's (13.0.1) latest build of Atmosphère (0.9.2).

sysmodules:
- emuiibo
- hid-mitm
- sys-clk
- sys-ftpd
- sys-netcheat
- ldn_mitm

homebrew:
- Kosmos Updater
- Kosmos Toolbox
- AmiiSwap
- Checkpoint
- Goldleaf
- NxThemes Installer
- Retroarch
- MelonDS

### Additional context?
I activated emuiibo, sys-clk, and sys-ftpd before the error.\
I'd guess that this is an emuiibo issue. 

[image]

j-james, 2019-07-08:
#595

SciresM, 2019-07-08:
sure doesn't sound like an atmosphere problem.

closed 2019-07-08 by SciresM