Atmosphere-NX/Atmosphere #629: Decrypt Package2 Header (Master Key Revision 6)

Muhammadmirza, 2019-07-15:
Updated to 8.1.0 using ChoidoujourNX from 6.2.0 CFW. Get Package2 Header decryption failed Master Key Revision 6. Any idea what I can do to fix this? 

Replaced SD card with ExFat and Fat32.
Updated Atmosphere to 0.9.2 
System will boot into recovery but will not boot into CFW/Stock.

designgears, 2019-07-15:
Are you starting fresh and deleting all of the old atmosphere files from the sdcard first? if not, try that.

Otherwise to me it sounds like you were using an older version of atmosphere that blocked ChoidoujourNX from fully flashing the firmware. If that is the case you'll need to manually update it with the Choidoujour (pc version)

Muhammadmirza, 2019-07-15:
Wouldnt It be better to downgrade first or just attempt an upgrade first?

designgears, 2019-07-15:
Doesn't matter either way, it's the same process.

Muhammadmirza, 2019-07-16:
Choidoujour PC steps says it doesnt support FW 6.2 to 8.1. Wouldnt that mean I have to downgrade first and then update.

Muhammadmirza, 2019-07-16:
Successfully downgraded back to 6.0.0 in the mean time.

SciresM, 2019-07-17:
Yeah, this is an issue where you ended up with wrong package2 flashed vs bootloader version -- using outdated choinx and/or atmosphere.

closed 2019-07-17 by SciresM