Atmosphere-NX/Atmosphere #631: Use ccache to improve compilation performance

fennectech, 2019-07-18:
## Feature Request

### What feature are you suggesting?
#### Overview:
- We can use ccache to improve performance significantly for users who have ccacne installed.</br>

#### Smaller Details: 
- Ccache caches source compilation meaning if a module doesnt change the system does not have to compile and can instaid pull from the cache making compilatiion siginificantly faster</br>

#### Nature of Request:  
- Addition
### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>
- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
This will make recompiling it atmo MUCH faster after compiling once.  </br>

friedkeenan, 2019-07-18:
Isn't it already that files that weren't modifed aren't recompiled?

fennectech, 2019-07-18:
It is isnt it :shrek:

closed 2019-07-18 by fennectech