Atmosphere-NX/Atmosphere #632: Feature Request: Config Generation

SunTheCourier, 2019-07-18:
## Feature Request

### What feature are you suggesting?
Generating Atmosphere's configuration files instead of packaging in release.<br>

#### Details: 
Somewhere during initializing of boot or of the config's component, the component should create the config file if it does not exist with default values.<br>

#### Nature of Request
- Addition
- Change

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Boot component or respective to each individual config's system module.<br>

### Why would this feature be useful?
Prevents users from accidentally reverting their configuration if it has been modified from default. </br>

SciresM, 2019-07-18:
Atmosphere explicitly has a policy of never writing configuration -- so far as it is concerned, configuration files are read-only.

SciresM, 2019-07-18:
There are plans to use default values instead of erroring when config isn't present, though, and moving default configuration to an examples folder (so that it wouldn't overwrite people's config any more).

closed 2019-07-18 by SciresM