Atmosphere-NX/Atmosphere #635: Failed to launch latest commit

jul2003, 2019-07-21:
## Bug Report

</br>
</br>

</br>
</br>
</br>
</br>
</br>

### What's the issue you encountered?

I git clone --recurse-submodules the atmosphere repo and run make && make dist
I successfully have atmosphere-0.9.2-master-d95e2095.zip and fusee-primary.bin in the out folder.
Unzip and copy the files to the sd card , add hbl and hbmenu in the correct folder of the sd card.
Launch atmosphere with the compiled fusee-primary.bin.
I do see the atmosphere logo but after black screen , i don't have the sept logo like when i use the release version.</br>
No change</br>
NA</br>

### How can the issue be reproduced?

</br>
clone the repo , compile and lauch</br>

### Crash Report

No crash report</br>
</br>
</br>

### System Firmware Version

4.1.0</br>
</br>
</br>
</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - Unofficial build
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment: NA

SciresM, 2019-07-21:
What's the last working commit for you? Please binary search and work back ^^

jul2003, 2019-07-21:
Commit 5bba0f4 works , commit fc7f06d doesn't . I took the compiled version from lavatech , i'm trying now to compile commit 20a48c3 but because i have an up to date devkitpro i think that's the reason i'm getting this error on compile
/Users/julien/code/Atmosphere-20a48c3a26c3f396d030e3ff6d531c1b6e8759a3/exosphere/src/configitem.c: In function 'configitem_get':
<command-line>: error: invalid suffix "x" on integer constant
/Users/julien/code/Atmosphere-20a48c3a26c3f396d030e3ff6d531c1b6e8759a3/exosphere/src/configitem.c:281:27: note: in expansion of macro 'ATMOSPHERE_RELEASE_VERSION_HASH'
             *p_outvalue = ATMOSPHERE_RELEASE_VERSION_HASH;
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [configitem.o] Error 1

jul2003, 2019-07-23:
@SciresM can you tell me what can i do to help to find how to fix this ?

hexkyz, 2019-07-23:
@jul2003 I've confirmed this issue and I'm currently tracking down what caused the regression. Thank you for testing!
Do you know of any other firmware versions that have the same issue? So far I've been able to confirm 1.0.0 to 5.0.0 are not booting.

jul2003, 2019-07-23:
I only have 1 device in 4.1.0 currently so i can't do test on other firmware

ZachyCatGames, 2019-07-23:
> @jul2003 I've confirmed this issue and I'm currently tracking down what caused the regression. Thank you for testing!
> Do you know of any other firmware versions that have the same issue? So far I've been able to confirm 1.0.0 to 5.0.0 are not booting.

Just tried 5.1.0-8.0.1. 5.1.0 is unable to boot with both hekate and fusee. 6.0.0-4.0 is unable to boot with fusee, but is able to boot with hekate (I think this is a different, unrelated issue). And 6.0.0-5.0 to 8.0.1 works fine

hexkyz, 2019-07-23:
I'm closing this issue as it should have now been fixed in the latest commit. I'll open it again if necessary.

closed 2019-07-23 by hexkyz