Atmosphere-NX/Atmosphere #636: error code 2001-0122 (0xf401)

stuie1, 2019-07-25:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### I was just done getting this you changed not freebird andever one called revered or something 1.01 I turned off the cpu hung it’s very easy it works fine but renix cfw I don’t get the news right but that cfw brings up Zelda game my problem is on atmosphere I have cheat codes they work fine with the Zelda 1.6 it used to work fine a week ago I haven’t don’t much different and now the game gives this error right when I start up game I used to get error if i wait longer then 1 sec before launching homebrew to go to cheats if I get in to game for more then 2 sec go to album then I get atmosphere error 1 sec I don’t get error now to main problem the game doesn’t start up on atmosphere on Zelda  ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ yes it does it over and over  ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ error code 2001-0122 (0xf401) ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System version

8.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### I’m using reswitched 

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2019-07-25:
This issue is incomprehensible. Please re-open using full english sentences, with punctuation.

SciresM, 2019-07-25:
You also failed to actually attach a crash report.

SciresM, 2019-07-25:
And failed to provide information about what version of atmosphere you're using.

closed 2019-07-25 by SciresM