Atmosphere-NX/Atmosphere #639: Just Dance Needs Fixes

InkxtheSquid, 2019-07-28:
## Bug Report

### What's the issue you encountered?

Just Dance 2018 and 2019 crash in countless places

### How can the issue be reproduced?

put a modified patch.ipk in the folder to see for yourself on what happens

### Crash Report

https://gist.github.com/InkxtheSquid/c3c134ffc052d22836e2c74bd8de7f63

### System Firmware Version

8.X.X</br>

### Environment?

- hekate
- both official latest and kosmos 
- yes
- yes

### Additional context?

- crashes in many places

piepie62, 2019-07-28:
That would be an issue with the patch, not Atmosphère

InkxtheSquid, 2019-07-28:
> That would be an issue with the patch, not Atmosphère

I need the patch file in order to make the mod work or else it might go bananas

ThatNerdyPikachu, 2019-07-28:
But it's still an issue with the patch, not Atmosphere.

SciresM, 2019-07-28:
Not an atmosphere issue. Talk to the developer of the patch.

closed 2019-07-28 by SciresM

InkxtheSquid, 2019-07-28:
I Made the patch its very important about it when it comes to modding Just Dance

InkxtheSquid, 2019-07-28:
> Not an atmosphere issue. Talk to the developer of the patch.

I made the patch file, its an important thing when it comes to modding Just Dance

ThatNerdyPikachu, 2019-07-28:
That doesn't mean it's an Atmosphere issue.

InkxtheSquid, 2019-07-28:
> That doesn't mean it's an Atmosphere issue.

look this used to work normally in the past so thus its an atmosphere issue, that people need to consider!

shchmue, 2019-07-28:
on which version of atmosphere did it stop working

ThatNerdyPikachu, 2019-07-28:
> look this used to work normally in the past so thus its an atmosphere issue, that people need to consider

you didn't mention that before now.

InkxtheSquid, 2019-07-28:
this has been happening to me ever since fw 7.6 got released and no one has been fixing this ever since

SciresM, 2019-07-28:
The crash report is a user break. An assertion is failing in the game.

If the issue doesn't happen without the patch, then it's an issue with the patch and not with atmosphere. It may be that your patch needs compatibility updates for newer firmwares.