Atmosphere-NX/Atmosphere #641: Atmosphere won't boot after upgrading to 8.1.0 (0.9.2)

tewcore, 2019-08-01:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ] Upgraded to firmware 8.1.0 using ChoidujourNX
[ Did you make any changes related to Atmosphère itself? ] No
[ If so, make sure to include details relating to what exactly you changed. ] N/A

### How can the issue be reproduced?

[ * ] Currently just try to boot into Atmosphere like normal.
[ Include a detailed step by step process for recreating your issue. ] Try to boot into Atmosphere like you usually would, and everything will come up like normal, until after the Nintendo Switch logo and Atmosphere comes up with the error code.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ] 
[01564687524_0100000000001000.log](https://github.com/Atmosphere-NX/Atmosphere/files/3458372/01564687524_0100000000001000.log)

[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

8.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ] Official
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: Don't think so.
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment: Didn't do this until after I upgraded to 8.1.0 using ChoidujourNX, was on 7.0.1 beforehand. 
- [ Any other information relevant to your issue. ]

SciresM, 2019-08-01:
Delete your custom theme. Not an atmosphere issue.

closed 2019-08-01 by SciresM

tewcore, 2019-08-01:
I deleted the themes off of the SD card and even deleted the .nro and it still persists.

ZachyCatGames, 2019-08-01:
> I deleted the themes off of the SD card and even deleted the .nro and it still persists.

Delete sdmmc:/atmosphere/titles/0100000000001000

SciresM, 2019-08-01:
Your custom theme lives in the folder ZachyCatGames mentioned, to be clear.

tewcore, 2019-08-01:
You guys are my heroes, thanks! ^-^