Atmosphere-NX/Atmosphere #642: Can't boot CFW after 8.1.0 update

kingafro000, 2019-08-02:
What's the issue you encountered?
Starting up Atmosphere only shows Error Message after Nintendo Switch Logo. (Error 2000-0000)

How can the issue be reproduced?
Update to 8.1.0(exfat) using ChoidujourNX and my own dump and then boot into CFW

Environment?

What bootloader (fusee, hekate, etc) was Atmosphère launched by: same result both hekate and fusee
Official release or unofficial build: Atmosphere 0.9.2
Do you have additional kips you're loading: no
Additional info about your environment: firmware version 8.1.0

SciresM, 2019-08-02:
Crash report? Deleted your custom theme?

stuie1, 2019-08-03:
Do u have a theme install delete titles for the damn themes or back everything up on sd card find a fat 32formatted and format sd card 2 32fat if u have a 500gb sd lower with out fat 32 will have errors best to be on fat 32

flightcoin, 2019-08-16:
Same here, how would you be able to clear out the themes after the update?

austin5623, 2019-08-16:
Literally just delete /atmosphere/titles/0100...100 (not 1000, that's mario odyssey)

flightcoin, 2019-08-16:
Works, thank for this.

shchmue, 2019-08-16:
note ...1000 is qlaunch, ...10000 is Mario Odyssey

closed 2019-11-21 by SciresM