Atmosphere-NX/Atmosphere #645: Autoboot into CFW after shutdown

kwanv923, 2019-08-03:
Hi, is there any setting to enable autoboot into CFW after a shutdown? Everytime i restart the device i will go to the Stock firmware and i cant get into Hetake menu unless i inject it with a pc again. Any clues on what i have done wrong during the installation?

austin5623, 2019-08-03:
1. You're supposed to use the issue template
2. That is impossible without exploits that will likely only ever be on 1.0-3.0, however you can reboot to a payload, useful for taking out your sd while in hekate or after a fatal crash

SciresM, 2019-08-03:
1) Does not use issue templates

2) After a shutdown, no Atmosphere code is running on your device and we lose execution. Since none of our code is running on your device and we have lost execution, we cannot do anything to regain execution without you once more performing the RCM exploit to get execution and run Atmosphere code.

tl;dr not possible.

closed 2019-08-03 by SciresM

stuie1, 2019-08-03:
Ya dude go on eBay and search hardmod Nintendo switch this lady will talk to u and I sent my switch to her just send it in a case with newspaper all in the box for safety she will install a trinket m0 it’s 69$ but your switch will restart like u want