Atmosphere-NX/Atmosphere #646: Atmosphere Not Loading After Fusee Payload Injection

richardsedai, 2019-08-04:
## Bug Report

### What's the issue you encountered?

Unable to get Atmosphere to load up. I've carefully followed all online instructions and YouTube videos on installing Atmosphere 0.9.2 but the software never takes hold and works. According to all the videos I've seen, as soon as I inject the fusee-primary.bin my Switch is supposed to automatically reboot and show me an Atmosphere loading screen. That never happens. My Switch stays in RCM mode no matter how many times I inject the fusee payload. I've tried redoing it with fresh downloads of Atmosphere and the included fusee-primary.bin in case my first files were corrupted but it still doesn't work.

### How can the issue be reproduced?

I don't know how it can be reproduced by your team. I'm just hoping you can advise me how to get the software to work for me or if I am doing something wrong.

### Crash Report

N/A can't access any crash reports since I can't get Atmosphere to load.

### System Firmware Version

8.1.0

### Environment?

- fusee-primary.bin that was included with Atmosphere 0.9.2
- Official release
  - Atmosphere 0.9.2
  

### Additional context?

Using TegraRcmGUI 2.6 on Windows 10. My Switch is an early unpatched model in the XAW10000000000 to XAW10074000000 range.

SciresM, 2019-08-04:
Well, one of the following must be true:

* Your unit is patched.
* There is a hardware issue with your sd card.
* There is an issue with the files on your SD card preventing fusee primary from loading fusee secondary.
* There is an issue with whatever process you're using to inject fusee ("you're doing it wrong").

Which seems most likely? I'm tempted to close this immediately since it seems so unlikely that this is an ams issue and not a user issue, and since support is given in the ReSwitched discord and not the github issues.

I'll leave it open though as a test -- if It's one of the above problems I will close issues like this one immediately in the future since that would indicate this is more of a user support problem.

SciresM, 2019-08-04:
Very good use of the issue template though, I appreciate it.

richardsedai, 2019-08-04:
You were right, there was an issue with the TegraRcmGUI program. I found a fusee-interfacee-tk injector program for Mac OS and it worked perfectly the first time. Atmosphere booted up immediately as it was supposed to. Thanks for responding so quickly. I'll close the ticket.

closed 2019-08-04 by richardsedai