Atmosphere-NX/Atmosphere #647: Not showing the cfw version on AMS 0.9.3

mightyburgerz, 2019-08-10:
## Bug Report

### What's the issue you encountered?

Updating my version 0.9.2 to 0.9.3 by extracting and pasting the new files and using the new fusee-primary to launch the atmosphere, but the cfw version was not able to show (only the firmware 8.1) but I can perfectly open homebrew without any errors. Tried to paste it again still the same.

### How can the issue be reproduced?

Boot to Atmosphere and the cfw version is not showing then themes are not able to install also

### Crash Report

N/A

### System Firmware Version

System Firmware Version 8.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: The new fusee-primary.bin
- Official release version 0.9.3 
 
### Additional context?

N/A

drrastovac, 2019-08-10:
You did something wrong. I also updated from 0.9.2 to 0.9.3 and it shows the current system version 8.1.0(AMS 0.9.3)S. Also, I installed a new theme without problems using the NXThemes Installer 2.0.1. Try deleting the Atmosphere, Sept and Switch(reboot_to_payload) files and put the new one. I did that, and I after that just put sigpatches in Atmosphere folder.

drrastovac, 2019-08-10:
I also updated from 0.9.2 to 0.9.3 and it shows the current system version 8.1.0(AMS 0.9.3)S. Also, I installed a new theme without problems using the NXThemes Installer 2.0.1. Try deleting the Atmosphere, Sept and Switch(reboot_to_payload), hbmenu.nro files and put the new ones.

Dark-Mind, 2019-08-11:
Try deleting all the files on sd card then only put the new files and boot

mightyburgerz, 2019-08-12:
@drrastovac got titles on the atmosphere folder is it alright?

mightyburgerz, 2019-08-12:
I do some trial and error and know the issue what makes it that way. It's either the kips or flags (Incognito) anyways removed the incognito and it works now ^_^
[image]

closed 2019-08-12 by mightyburgerz