Atmosphere-NX/Atmosphere #650: Black Screen on 9.3

takashi1kun, 2019-08-18:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I updated to atmosphere 9.3 and now i get a black screen after the third logo, the black and white one, first comes up the sept one, then the atmosphere one and then the black and white one then black screen, 

### How can the issue be reproduced?

### Crash Report

### System Firmware Version

I don't remember, there has been some time science i used the switch 

### Environment?

- i used hekate to sideload the fusee primary payload
-i use emuemmc
-FAT32
-emuemmc is correctly configured

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

lwouis, 2019-08-19:
Same experience here. I'm on 7.0.0. I tried replacing atmosphere on the SD card with 0.9.3, 0.9.2, 0.9.1 but I always get a black screen after the second logo. Then I have to hold power for around 10s, then I can reboot in normal mode or RCM.

I'm now wondering if I should upgrade to 8.1.0 and try 0.9.3 again

takashi1kun, 2019-08-19:
> **Update:** Turns out the issue was that the SD card was formatted as exFAT and it seems Atmosphere requires FAT32 (at least for OS 7.0.0 which my Switch is running). I formatted to FAT32, reinstalled the Atmosphere 0.9.3 and I'm past the black screen now
Weird i have FAT32 not exFat, i even reformated the SD and still black screen

hexkyz, 2019-08-19:
@takashi1kun Does it happen without hekate too (launching only with fusee-primary instead of chainloading)?

takashi1kun, 2019-08-19:
> @takashi1kun Does it happen without hekate too (launching only with fusee-primary instead of chainloading)?

Just tried, yes it still happens directly injecting fusee primary whitout chainloading

Resaec, 2019-08-24:
If you had themes installed that broke on the update, those could prevent you from getting to the lockscreen.

Did you replace the old atmosphere files on your SD or did you delete them beforehand?
You should always delete all old files, copy over the new ones (there should be no replace dialog) and edit the config files to fit your needs again.

takashi1kun, 2019-08-26:
> If you had themes installed that broke on the update, those could prevent you from getting to the lockscreen.
>
> Did you replace the old atmosphere files on your SD or did you delete them beforehand? [...]

I did replace the whole atomosphere files several times, no luck

stuie1, 2019-09-03:
If none of this worked for u try booting up sx os I did a issue labeled at the beginning of the page hey if u used backed up files to switch and doesn’t work u issue will be closed yes I did use different boot files and it skrewed up my sx os but I sent my system telling this lady on eBay ab9ut what I did and she moved me over to atmosphere idk what she did but I don’t use sx os anymore on that system I use atmosphere or anything else sx os always boots up to a black screen ::/ this is all I can help with and if u want to contact the lady that fixed my switch she might have 2 more trinket m0 to install to switch so u don’t have to use a third party payload injector but that lady saved my switch the only issue is I didn’t understand that I had themes installed and I messed up software with older boot files 0 and 1 I guess my system never made the full back up that time skreqed my sx os but worked out in a way.... need help with contacting that lady app u do is ship your system to her the safest way possible and she ships it back super fast lmk sbaile6492@gmail.com

Hay1tsme, 2019-09-11:
> 
> 
> I have had this issue as well and was assuming it was because my Switch OS is on 9.0.0. Is it a bug in Atmosphere or does it not work for my OS version?

Atmosphere doesn't yet work on 9.0.0. If you updated you'll need to wait for a new version of Atmosphere.

richardsedai, 2019-09-11:
Thanks for the info. There was no mention on the download page regarding what version it works with. Hope they add that to future versions so we know.

takashi1kun, 2019-09-11:
No, my SwitchOS is definitively not on 9.0.0 i always update manually and last update was 8. something

meeces2911, 2019-09-12:
@takashi1kun I had something similar when I first updated atmosphere. I found loading the latest version of Hekate, powering off your switch, and then loading atmosphere worked for me.

closed 2019-09-16 by SciresM

takashi1kun, 2019-09-21:
@SciresM Still happens on 0.9.4
@meeces2911 Tried that, did not worked.

SciresM, 2019-09-21:
@takashi1kun Are any crash reports generated? What's inside your /atmosphere/titles directory?

takashi1kun, 2019-09-21:
@SciresM I solved it, i belive that the emunand partition go corrupt, because i just wiped it and remake it from the original rawnand and boot files i used to make it in the first place and now it boots correctly.