Atmosphere-NX/Atmosphere #651: Support for >4GB romfs.bin files on FAT32

OatmealDome, 2019-08-18:
## Feature Request

### What feature are you suggesting?
#### Overview:
- The romfs of some games can exceed 4GB in size. For example, Splatoon 2's romfs with the 3.1.0 update is approx. 4.9GB large.
- A romfs.bin file for these games cannot be stored on FAT32 partitions.

#### Smaller Details: 
- [Related TODO in ``fsmitm_service.cpp``?](https://github.com/Atmosphere-NX/Atmosphere/blob/fc7f06dc78760d90313c76a4b07dc9966a5e3bc6/stratosphere/ams_mitm/source/fs_mitm/fsmitm_service.cpp#L369)
- Possible file layouts:
    - ``romfs.bin``, ``romfs1.bin``, etc?
    - ``raw_romfs/00``, ``raw_romfs/01``, etc?

#### Nature of Request:  
- Change

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)
    - fs.mitm

### Why would this feature be useful?
Would allow users to avoid the buggy Horizon exFAT driver when working with large games.

fincs, 2019-08-18:
Just use Horizon OS's large file support (i.e. folder with archive bit set, and files inside named 0, 1, 2, etc)

OatmealDome, 2019-08-18:
I had no idea that was a thing. Thanks.

closed 2019-08-18 by OatmealDome

SciresM, 2019-08-18:
@OatmealDome libnx got support for creating these files with fsdevCreateFile recently, too.

shchmue, 2019-08-18:
Have you tried making a standard archive? a folder named romfs.bin with archive bit set that contains splits named 00, 01, etc of max size 4294901760 bytes

ExxonDivinely, 2019-08-19:
how do unscrib from emale

jakibaki, 2019-08-19:
@ExxonDivinely click on the unwatch button on https://github.com/Atmosphere-NX/Atmosphere