Atmosphere-NX/Atmosphere #652: update 8.1 instead of 8.1exfat now BLACK SCREEN

jrme421, 2019-09-04:
We would like to use GitHub to keep track of problems/feature requests.  
If you have a question, please join the ReSwitched discord for help. 
- Discord link: https://discordapp.com/invite/DThbZ7z 

like the title say, after i used choixdujour, i selected just 8.1 instead of 8.1 exfat, choix du jour showedd my old versoin only after the update, it was 7.1exfat now i cannot boot atmosphere

any help

HookedBehemoth, 2019-09-04:
Format your SD card to FAT32

jrme421, 2019-09-04:
is that the only way? its about 460gb, plus once i do that, how to get back to exfat?

jakibaki, 2019-09-04:
Don't use exfat, the driver is buggy af on the switch so at some point (probably soon) it would corrupt.

You can format big sd-cards to fat32 just fine.

HookedBehemoth, 2019-09-04:
if you REALLY want exFAT you can boot OFW with an exfat card and let it download the exFAT update again.

jrme421, 2019-09-04:
my sd card is in but it wont boot, its a black screen

jrme421, 2019-09-04:
my sd card is fat 32

jrme421, 2019-09-04:
ok my bad, it might another problem i launched from the computer with heckate and from there i choose atmosphere but black screen, then i try launch cfw emunand et systemnand but black screen as well aft sept logo

HookedBehemoth, 2019-09-04:
If you didn't install the exFAT update on emuMMC you have to use a FAT32 SD card because you can't boot that with OFW.

jrme421, 2019-09-04:
solved, i had to copy the kosmos folder with everything again, no idea why

tahnk everyone

closed 2019-09-04 by SciresM