Atmosphere-NX/Atmosphere #654: Korean patch does not working.

bsredsox, 2019-09-06:
Hi My name is Kim and I'm very pleased to use your marvelous achievement.
Unfortunately, I have an issue about Korean patch.

Once I copied the Korean patch files into titles folder, I expected to enjoy converted Korean things but patch files did not work unfortunately.

I do not know what is going on thus I googled this issue and found out Korean patch do not valid on 0.9.3.

Could you please consider this issue once you have update plan?

Thank you and have a good day.

SciresM, 2019-09-06:
Hello, 

I'm afraid I don't quite know what you mean by "Korean patch files". Could you explain exactly what you are trying to do?

bsredsox, 2019-09-06:
Thank you for the comment.
For example, Some talented Korean users made Korean patch files for Monster Hunter (Japanese version) as most users do not understand Japanese language.

Generally, we copy these patch files to title folder like Sdcard/atmosphere/titles/
After copy of patch files, we can enjoy converted game on mother language.

However Atmosphere 0.9.3 does not support this patch files so we are in very serious trouble sadly. Not only Monster hunter, Korean users made several patches for games and now they are making patch for Xenoblade chronicles 2 currently.

We googled this issue and found same agony and situation, we really admire this tragedy will be solved asap.

Thank you.

uaevuon, 2019-09-06:
No. It works like a charm. 
You meant [this](https://blog.naver.com/dudaos1123/221390808374), right? It works on AMS 0.9.3.

Make sure you have clean and not corrupted atmosphere files on the sdcard.
Archive bit fix in hekate might help.

bsredsox, 2019-09-06:
Thank you, i will try it.

mixtoism, 2019-09-25:
should be fixed by now?

closed 2019-09-27 by bsredsox