Atmosphere-NX/Atmosphere #657: Nintendo switch v.9.0

advenger005, 2019-09-14:
Me gustaría saber cuando saldrá la nueva versión de Atmosphere para la nueva versión 9.0? Me actualizaron la switch por accidente y ninguno de los patch me funciona se queda la pantalla en negro😓😓

D3fau4, 2019-09-14:
Eta 5h

fincs, 2019-09-14:
Por favor rogamos paciencia. Estamos todavía poniendo todo a punto para las nuevas versiones de Atmosphère y libnx.

> OP said:
> I'd like to know when will the next version of Atmosphère come out for the new {system} version 9.0. My Switch got updated by accident and none of the patches work for me, screen stays black 😓😓
>
> I replied:
> Please be patient. We are still getting everything ready for the next versions of Atmosphère and libnx.

closed 2019-09-14 by fincs