Atmosphere-NX/Atmosphere #660: 0.9.4 crashes the switch

Axodique, 2019-09-14:
## Bug Report

### What's the issue you encountered?

I replaced hekate and atmosphere to the latest versions in the file (hekate or not it crashes) and it crashes just after the nintendo switch screen
### How can the issue be reproduced?

### Crash Report

https://gist.github.com/Axodique/3a27075a8dd348a9e03f64e4a52efe2f

### System Firmware Version

9.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

tegra rcm Environment

Axodique, 2019-09-14:
UPDATE: I tried with just atmosphere and it worked. i'll try putting homebrew one by one to find the one that crashes me.

SciresM, 2019-09-14:
Cool, not an ams issue then.

closed 2019-09-14 by SciresM

friedkeenan, 2019-09-14:
Might be because of https://github.com/jakibaki/hid-mitm

Axodique, 2019-09-14:
It was actually the nintendo folder. people who installed games on their sd card ilegally be careful

Axodique, 2019-09-14:
Actually,i was with the fact that i was just replacing the files and not reseting the atmosphere folder to 0.

Axodique, 2019-09-14:
Last update i swear! it was the titles folder.

Shenron0, 2019-09-15:
For me it just freezes on the Nintendo Switch logo