Atmosphere-NX/Atmosphere #661: Is there any way we could make the apps/games installed on ReiNX to work with Atmosphere after a transition?

dothackjhe, 2019-09-15:
So, basically, I was informed that the transition would be as simple as deleting the files on the micro-SD Card, with the exception of the Nintendo folder where the data for the installed apps/games is stored. 

Putting that in mind, I made a backup of the "Nintendo" from an old micro-SD Card which is used to run ReiNX, and transferred to a new and bigger-capacity micro-SD Card, which runs on Atmosphere/Kosmos. 

The thing is, while the systems sees the apps/games I had previous installed when was with ReiNX, it would not run it, which gives off this error: "The software is not currently on the console. Please redownload the software to continue. If the software was originally downloaded to a micro-SD Card, please cancel, turn off the console, and insert that micro-SD Card."

Is there a way I could run the apps/games I had installed with ReiNX on Atmosphere/Kosmos?

hurrian, 2019-09-15:
Did you purchase the games from eShop?

dothackjhe, 2019-09-15:
Ohh, right. Sorry, I forgot that the team behind Atmosphere does not support the idea of pir@cy.

fincs, 2019-09-15:
Piracy is *not* supported.

closed 2019-09-15 by fincs