Atmosphere-NX/Atmosphere #662: Fatal error faild to start stage 2

albatros2808, 2019-09-15:
Hi,
Im having problem while im trying to fuse atmosphere on my nintendo switch my switch is on fw 8.0.1 and im using sandisk 128gb microsd card that i have formated to exfat ive tried to hack my switch according to a guide that ive seen online but when im trying to inject fusee i get the fatal error message maybe you know what is the problem or maybe you can give me another guide so i can try again?
Thanks in advance for your help!
We would like to use GitHub to keep track of problems/feature requests.  
If you have a question, please join the ReSwitched discord for help. 
- Discord link: https://discordapp.com/invite/DThbZ7z

SciresM, 2019-09-16:
Seek help in user-support, or else use the issue templates properly...

closed 2019-09-16 by SciresM

stuie1, 2019-09-17:
Why are u formatted to exfat u have to find a custom software it’s a randomly small program I don’t remember the name thou but your looking for a fat32 formater u need to formate your sd to fat32 skrew exfat that shit doesn’t work 

Sent from Stuart Bailey Jr