Atmosphere-NX/Atmosphere #667: Switch does not read cartridges after update via Atmosphere

Lorenzooone, 2019-09-18:
## Bug Report

### What's the issue you encountered?

After updating my Switch to the latest version, it cannot read cartridges anymore.</br>

### How can the issue be reproduced?

I updated my Switch to the latest version (via Update button with Atmosphere running in the background) and I updated Atmosphere too in order to use said Switch.</br>
After that, the Switch could not read cartridges anymore.
I tried formatting the SD and putting back on it a clean atmosphere version. Didn't work.</br>
After that, I restored to a previous Firmware Version I had a backup of. (7.0.0) The cartridges were read properly. I tried re-updating and the same issue happened.

### System Firmware Version

9.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary.bin
- Official release or unofficial build: Official Build 0.9.4-master-a750e55f
- Do you have additional kips or sysmodules you're loading: No.

ZachyCatGames, 2019-09-18:
Set nogc to 0 in BCT.ini. Do note that if you do this you will burn a fuse in your cart reader and never be able to use it on <9.0.0

d4mation, 2019-09-18:
If you want to play cartridges on 9.0.0, you'll need to allow the gamecart controller firmware to update. Atmosphere prevents this by default.

From the release notes:

https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.9.4

> **Please note**: 9.0.0 updated the switch's gamecart controller ("lotus") firmware.
> * To prevent the burning of lotus fuses, Atmosphere now applies nogc patches when booting into > 9.0.0 with fuses below 9.0.0.
> * If you wish to play gamecarts on 9.0.0 (but forever prevent yourself from playing them on < 9.0.0), set nogc = 0 in BCT.ini.

Lorenzooone, 2019-09-18:
Oh, okay. Thanks for the quick answers! I'll see what to do with that later.

closed 2019-09-18 by Lorenzooone

anaxxo, 2021-02-06:
@Lorenzooone Hi, I'm using fusee primary with atmosphere 0.17.1 and FW 11.0.1 and I'm getting the card error, wondering how did you fix this issue ? If any information you could share it'll be great, thanks

Lorenzooone, 2021-02-06:
There should be a BTC.ini (or config.ini) file in the atmosphere/config_templates folder. 
In that file, add, below "[stratosphere]" a line that says:
nogc = 0

NaturalHarmonia, 2021-02-06:
I did that and the game cart still doesn't work for me. I've been having this problem for months. My controllers are updated as well.

SciresM, 2021-02-06:
Only under /config. /config_templates is for templates, you don't edit files in there.

inkk50, 2021-02-09:
> Only under /config. /config_templates is for templates, you don't edit files in there.

Can you explain me in a deitaled way pls or with a photo? I tried everything and it didn't work

jnilsson, 2021-02-16:
If you boot via Hektate make sure you go to options in the hektate bootloader and toggle nogc off there. This worked for me.

LordGorzul, 2021-02-20:
> Only under /config. /config_templates is for templates, you don't edit files in there.

I dont have any files under /config folder is empty. Ill try copying it from config_templates