Atmosphere-NX/Atmosphere #668: Blank screen HELP!!

chubbysocks, 2019-09-19:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

chubbysocks, 2019-09-19:
I tried to manually update my firmware from 8.0.1 to 8.1.0 using ChoiDujourNX. I selected reboot but nothing happened, just a blank screen.

Mika-Westphal, 2019-09-19:
Atmosphere 0.8.8 isn't supporting 8.1.0. Please update to the newest Atmosphere version.

closed 2019-09-19 by fincs

chubbysocks, 2019-09-20:
> Atmosphere 0.8.8 isn't supporting 8.1.0. Please update to the newest Atmosphere version.

I just drag and dropped the latest atm on my sd card. TegraRcmGUI doesn't recognise the switch.

Mika-Westphal, 2019-09-20:
so you already tried a USB port at the back side of your computer?

chubbysocks, 2019-09-20:
> do you already tried a USB port at the back side of your computer?

Ok...I reverted back to atm 0.8.8 and used the usb port from the back off my desktop, and a little message popped up for a few seconds (this hasn't happened for a few days now). When I inject the payload nothing happens. I tried it with the newest atm 0.9.4 and tegraRcmGUI doesn't connect to my switch. 

Am I doing something wrong??? with the atm 0.9.4 I'm using TegraRcmGUI 2.6 (installed driver), it doesn't connect to my switch.

chubbysocks, 2019-09-20:
Ok... I think I might have made a break through. I tried atm 0.9.0 (no RCM connection), then tried 0.9.1.... the atmosphere logo just flashed up :) but the battery was in the red, when I go to disconnect the usb c cable the console dies. I'm going to see if my PC can charge the switch with enough power so I can switch the power outlet to charge it up.

chubbysocks, 2019-09-20:
> do you already tried a USB port at the back side of your computer?

Ok, I think that was the problem. The battery was dead and using the usb port at the back gave it enough power. I'm now on 8.1.0/0.9.4, thanks. I had to use this method to charge the switch

https://www.youtube.com/watch?v=Am9Y__UxFZ4

This is the last question. Now I've upgraded, my games won't work when I click on them. Is this an easy fix?

chubbysocks, 2019-09-20:
Ok I found  them here https://gbatemp.net/threads/i-heard-that-you-guys-need-some-sweet-patches-for-atmosphere.521164/