Atmosphere-NX/Atmosphere #670: Nintendo Switch won't boot after downgrading from 9.0.0 to 8.1.0

Pizza9555, 2019-09-20:
Hello, I am new to the Nintendo Switch hacking scene and love all the hard work and dedication you all have been putting towards this system! But after i bought a switch today (Firmware version 9.0.0) i dove straight into watching videos and tutorials on how to hack it. I previously bought a jig on amazon to arrive today when i purchased the system as well as a 400GB micro sd card. After it arrived i put the system into RCM mode, downloaded TegraRcmGUI v2.6 and atmosphere 0.9.4, and injected the fusee-primary.bin Got into CFW, went into the Album holding the R button to find i could now run homebrews! I then downloaded ChoiDujourNXv102 and downgraded to OFW 8.1.0 so that i could run other things that didn't yet support OFW 9.0.0 and upon booting back up, i got this error code. I've tried for hours injecting different payloads, trying to use different versions of Hekate, ReiNX, atmosphere 9.3, 9.2, 9.1, etc. and cannot for the life of me get this thing to boot up past one of these error screens :/ now of course i did not back up my NAND or System prior to downgrading the firmware... can you please think of anything that i can do?? I disabled autoRCM but it still wont even try booting into Recovery mode for me to install newer firmware or anything. I'm all out of options. Is there any hope for my Switch? 

Thank you for your time.
[image]

SciresM, 2019-09-20:
1) Use issue templates in the future, please.

2) The save file version for NS is too new for 8.1.0. You'll need to delete the save manually, or else restore a backup. This happens when downgrading across major system versions most of the time.

closed 2019-09-20 by SciresM

stuie1, 2019-09-20:
Well I had this similar issue when I thought I made a back up and I just restored boot files 1 and 0 from a different version and i don’t get a error message it just never bolts into sx os I sent my Nintendo switch to a girl who install trinket m0 on the switch so u don’t have to plug in a stupid payload injector I told her what I did and she some how fixed my system and saved all my data I was really scared but I can message her find out if she has anymore trinket m0 and she can possible fix your switch but ughhhhh the other problem u might be having is that u installed stupid themes when u install themes then update u have to uninstall them try to delete the damn files from your Sd card that are in the titles folder in the certain payload file u have injected to switch delete files that are not games and look starve delete them

Sent from Stuart Bailey Jr