Atmosphere-NX/Atmosphere #671: Sysnand modified/updated when using ChoidujourNX in EMUMMC

manolz1, 2019-09-21:
### What's the issue you encountered?

I had EMUMMC setup for CFW (8.1.0) and Sysnand (9.0.0) as a clean untouched OFW.
I updated AMS from 0.9.3 to 0.9.4 by grabbing the files from the github release page. I ran ChoidujourNX in EMUMMC to update my EMUMMC from Horizon 8.1.0 to 9.0.0. Update was successful and worked (for the EMUMMC, didn't test Sysnand post upgrade). Was experiencing some gamecart issues in CFW EMUMMC (gamecart error kept popping up) so I just decided to downgrade my EMUMMC back to horizon 8.1.0. I re-ran ChoidujourNX and downgraded to 8.1.0.

Sysnand which was 9.0.0 (It *may* have actually been 8.1.0, the reason I believe that it was 9.0.0 is because it's always online and was used to play fortnite recently, either way it is now corrupted by actions performed in EMUMMC) is now reporting it's 8.1.0 (from the error message below) and is now failing to boot with official Nintendo error message:

> Error Code: 2002-2634
> 
> An error has occurred.
> 
> Please press the POWER Button and restart the console. If you are unable to restart the console, hold the POWER button for 12 seconds to turn the console off.
> 
> If the problem persists, refer to the Nintendo support website.
> 
> support.nintendo.com/switch/error
> ______________________________________________________________________
> (X)1 (XAW4xxxxxxxxxx)
> (X2) 8.1.0

CFW EMUMMC is also reporting 8.1.0. Booting into CFW EMUMMC also encounters an AMS crash with the same error code. I tried downgrading to AMS 0.9.3 but still the same errors.

[ Did you make any changes related to Atmosphère itself? ]</br>
No

### How can the issue be reproduced? 
See above

### Crash Report
https://gist.github.com/manolz1/141f11f5f5740b42c1ee53f38441c060

### System Firmware Version
9.0.0/8.1.0

### Environment?
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe primary via hekate 5.0.2
- Official release or unofficial build: Official AMS 0.9.4 directly from github release page
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A

SciresM, 2019-09-21:
From the 0.9.4 Release notes:

* **Please note**: 9.0.0 updated the switch's gamecart controller ("lotus") firmware.
  * To prevent the burning of lotus fuses, Atmosphere now applies nogc patches when booting into 9.0.0 with fuses below 9.0.0.
  * If you wish to play gamecarts on 9.0.0 (but forever prevent yourself from playing them on < 9.0.0), set `nogc = 0` in BCT.ini.

You played a gamecart on 9.0.0, and thus burnt your lotus fuses. You will never be able to use a gamecart on lower than 9.0.0, ever again. Remove your inserted gamecart to fix the error, or set `nogc = 1` in your BCT.ini to forcefully enable nogc on 8.1.0 (this will fix the error code, but you will still be unable to use cartridges).

closed 2019-09-21 by SciresM

manolz1, 2019-09-21:
Yes that’s fine, my plan was to be on 9.0.0 for everything anyway. 

The problem I raised is that my Sysnand got modified and corrupted from operations performed inside EMUMMC. Everything else was just the background information to be as descriptive as possible. 

Thanks

SciresM, 2019-09-21:
Your sysnand did not get corrupted at all. You burnt lotus fuses, which causes the error.

manolz1, 2019-09-21:
Thanks for the quick replies, appreciate it.

You're right. I can boot into CFW by forcing nogc = 1 with AMS, and have updated my EMUMMC back to 9.0.0

I guess I have to risk booting into OFW sysnand with AMS on, to update OFW to 9.0.0 now too.

Thanks SciresM, appreciate your time.