Atmosphere-NX/Atmosphere #672: Freezes on Nintendo Switch Screen on 9.4.0

Nomercyman1, 2019-09-23:
## Bug Report

[ * ]

### What's the issue you encountered?

[ Injected fusee primary on the Switch, to boot into homebrew. It loaded up the Atmosphere and Sept splash screens, but seems to freeze at the Nintendo Switch splash screen.   ]</br>
[ I changed the fusee primary used to boot into homebrew to the latest version, and replaced all the 9.3.0 files with those for 9.4.0, since I updated to software version 9.0.0 ]</br>
[ updated files from "atmosphere-0.9.4-master-a750e55f+hbl-2.2+hbmenu-3.1.1.zip"
and "fusee-primary.bin" ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Inject the "fusee-primary.bin" to load homebrew ]</br>

### Crash Report

[ N/A]</br>
[ N/A ]</br>

### System Firmware Version

9.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version 9.4.4 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ N/A ]

DanielC-08, 2019-10-01:
Do you mean 0.9.4 atmosphere and 9.0.0 horizon os?

SciresM, 2019-10-01:
Please remove your custom system modules/custom theme. If you have files under atmosphere/titles/0100000000001000, delete them. If that doesn't work, please list the folders under /atmosphere/titles to be told what to delete.

Nomercyman1, 2019-10-04:
9.0.0 switch software. But I realized it was the Game Cartridge in the slot
causing the problem. Though it only seemed to be an issue on the first boot
up into Homebrew. Now, working fine with game cartridge inserted.

closed 2019-10-06 by SciresM