Atmosphere-NX/Atmosphere #675: Save Data corruption while using CFW

crisftw, 2019-10-08:
Latest version of Atmosphere causes an issue to a game called Castle Crashers (Not pirated). The game run well. No crashes. The one issue that is being reported around is that after making progress and quitting the application, it will delete character progress. Running OFW, the game will not delete character progress after starting the game up. All this was done on 8.1.0, 9.0.0 and 9.0.1.

SciresM, 2019-10-08:
In the future, please use issue templates.

What happens if you enable the experimental save redirection to SD for the game?

xPrinny, 2019-10-08:
I'll write a bug report.

## Bug Report

In the game Castle Crashers, save data does not get loaded properly after loading or switching profiles in game. (Loading a second profile for a second player works fine.)</br>

### What's the issue you encountered?

Save data does not get loaded properly after loading or switching profiles.</br>

### How can the issue be reproduced?

Launch the game and load a profile.</br>
Play the game normally, exit to map to save data.</br>
After game is closed and reopened or profile is switched save will not be loaded properly.</br>

### Crash Report

N/A

### System Firmware Version

8.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: 
  - Unofficial build
  - Komos distruption of Atmosphère (0.9.3)
- Do you have additional kips or sysmodules you're loading: hid_mitm, ldn_mitm
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment: N/A
- As for mentions about it deleting progress is mostly false. Game does not load data properly and trying to play will overwite it with current playthrough.
- Experimental save redirection to SD does nothing. (I might have not done this right.)

crisftw, 2019-10-08:
My apologies SciresM. I have enabled the experimental save data redirection to the SD but it doesn't seem to work for me. I got support to see if I enabled it properly but it didn't work in the discord server. I also wanted to use older versions of Atmosphere and I only got down to 0.9.1 and nothing worked. Character progress was still deleted

ItsCinnabar, 2019-10-09:
For what it's worth, this exact process appears to allow the game to properly save while under atmosphere : 
https://streamable.com/ptiur
https://gbatemp.net/threads/castle-crasher-remastered-player-1-save-issue.548517/#post-8800872

xPrinny, 2019-10-09:
> 
> 
> For what it's worth, this exact process appears to allow the game to properly save while under atmosphere :
> https://streamable.com/ptiur
> https://gbatemp.net/threads/castle-crasher-remastered-player-1-save-issue.548517/#post-8800872

Oh, that's my video. It actually allows it to load properly not save properly. I think I forgot to mention that achievement do get loaded when you pick your profile but not character data while that mini-game somehow makes the data load properly.

Saving does not seem to be the issue, it's loading. You can prove this by playing and saving normally, reload the game with a different profile or swap profile in game, then pick the first profile as player 2 and you can see it loads properly.

crisftw, 2019-10-09:
🤔 Well if the game doesn't load properly due to the CFW and it has a solution, then I guess there isn't much SciresM can do as the issue is unknown. But I can tell you by experience that SXOS 2.8 worked fine with this game until you update to 2.9. Its been proven SXOS is using Atmosphere source. I just don't want many other games to bug out because of CFW

ItsCinnabar, 2019-10-09:
Just because someone found a work around doesn't mean it can't be fixed atmosphere side. I only posted that to hopefully give a bit of a starting point to the issue, so it can been seen what changes from normal loading vs post workaround loading.

SciresM, 2019-11-21:
Is this still a problem?

crisftw, 2019-11-24:
@SciresM Yes it is

reminon, 2019-12-22:
I'm not understanding the issue you are having. I'm on 9.0.1, with ams 0.10.1, and I've had no save/load issues in either single player or co-op.

crisftw, 2019-12-22:
> I'm not understanding the issue you are having. I'm on 9.0.1, with ams 0.10.1, and I've had no save/load issues in either single player or co-op.

That's because it seems like the latest update of Atmosphere fixed the issue

reminon, 2019-12-22:
My bad, I'm an idiot. Sorry for the intrusion. Lol

closed 2019-12-30 by SciresM