Atmosphere-NX/Atmosphere #681: Specific files become corrupted

MegaSuperab, 2019-10-15:
## Bug Report

### What's the issue you encountered?
When booting up the latest version of Atmosphere, specific files on the microSD card become corrupted. These files are, among others:
- Specific homebrew apps, such as Ultimate Mod Manager and Retroarch
- Any snapshots/videos taken with Atmosphere before booting
- Any mods for Super Mario Maker 2 (01009B90006DC800) (these only sometimes corrupts)

### How can the issue be reproduced?
To corrupt files:
1. Boot up Atmosphere
2. Open Homebrew launcher, then open an app that can open an FTP client
3. Send a specific file through the client onto the SD card
4. Reboot Atmosphere
5. Wait a few minutes
6. Files should be corrupted
To crash Atmosphere, simply attempt to load a corrupted SMM2 mod or open a corrupted homebrew app.

### Crash Report
https://gist.github.com/MegaSuperab/1cc4965f67b52f9da493050270a171a0

### System Firmware Version
9.0.1

### Environment?
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Checkpoint, ftpd, Lockpick, NX-Shell, NXThemesInstaller, reboot_to_payload, retroarch_switch (Corrupts), tetriswitch, Ultimate_Mod_Manager (Corrupts), DDLC-LOVE.Switch, DeSmuME-NX, EdiZon, melonds, nxdumptool, pinballnx, smw (Super Mario War), Snake, vgedit

### Additional context?
- Additional info about your environment: The microSD card is formatted to exFAT.

hippydave, 2019-10-15:
> * Additional info about your environment: The microSD card is formatted to exFAT.
Don't use exfat for homebrew. Problem solved.

MegaSuperab, 2019-10-15:
> > * Additional info about your environment: The microSD card is formatted to exFAT.
> 
> Don't use exfat for homebrew. Problem solved.

Please give me factual evidence that exFAT is what’s causing the problem. If you can’t, you aren’t helping me.

fincs, 2019-10-15:
There's countless other people, way too many to list here, that have had SD card corruption problems solely caused by the incredibly subpar and dangerous exFAT driver Nintendo uses in Horizon. It is a common advice to avoid using exFAT, and switch to a SD card formatted with FAT32. This is a necessary troubleshooting step that we need you to make, in order to figure out if you are indeed yet another victim of exFAT, or if on the other hand there's an actual unrelated issue going on in your system.

naikrovek, 2019-10-15:
> Please give me factual evidence that exFAT is what’s causing the problem. If you can’t, you aren’t helping me.

*You're the one asking for help.*  Take the advice given, if you want to see for yourself.  Or, don't, and continue to suffer file corruption.  Up to you.

shchmue, 2019-10-15:
test the same procedures on fat32 on a different card if you must, but you'll find that homebrew and mods that are up to date and compatible with the firmware they're launched from do not cause this behavior

other things can cause album crashes like the dump you linked though, like launching homebrew that requires lots of ram (retroarch and hb-appstore for example) using album override instead of title override

MegaSuperab, 2019-10-16:
> > Please give me factual evidence that exFAT is what’s causing the problem. If you can’t, you aren’t helping me.
> 
> _You're the one asking for help._ Take the advice given, if you want to see for yourself. Or, don't, and continue to suffer file corruption. Up to you.

Well maybe if he answered the question _nicely_ I wouldn't be so rude.

closed 2019-10-16 by MegaSuperab