Atmosphere-NX/Atmosphere #684: my problems with 9.0.1 and atmosphere 0.9.4

hmax42, 2019-10-19:
## Bug Report

### What's the issue you encountered?

- Witcher 3 aborts during the intro before the first video is played. it shows a popup with the text "The software was closed because an error occured." and a close button.
- Darkside Detective & Voez ignore joycon input, only touch works

### How can the issue be reproduced?

start the games from cart with active atmosphere 0.9.4 under 9.0.1

### Crash Report

it's not a crash</br>

### System Firmware Version

9.0.1 with atmo 0.9.4</br>

### Environment?

- Tested direct boot with atmosphere's fusee-primary and boot via hekate 5.0.2
- all 3 games work with no problems without atmosphere in official firmware 9.0.1
- Do you have additional kips or sysmodules you're loading: NO (tested with a fresh atmosphere installation and the 2.0.0-9.0.1 sigpatches) 
- Homebrew software installed: nothing running in the background, just installers/managements tools (lithium, goldleaf, checkpoint), cannonball-nx. nxmtp , ftpd, retroarch

### Additional context?

- also happens without the sigpatches for 9.0.1
- nogc is disabled

SciresM, 2019-10-19:
* "Darkside Detective & Voez ignore joycon input, only touch works"

You probably have custom sysmodules causing this. What folders do you have under /atmosphere/titles?

* Witcher 3 aborts

This is a crash. Please upload the crash report.

hmax42, 2019-10-19:
ok, here are my crash reports

[crash_reports_witcher3.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3746596/crash_reports_witcher3.zip)

want dumps too?

these are the content my titles directory
010000000000000D
0100000000000032
0100000000000034
0100000000000036
0100000000000037
01006C900CC60000

the last one's just an audio patch, i tried without and no change

SciresM, 2019-10-19:
Could you also list the content of your atmosphere/kips directory, if you have one?

hmax42, 2019-10-19:
- default_nogc
- fs_patches

SciresM, 2019-10-19:
atmosphere/kips, not atmosphere/kip_patches.

hmax42, 2019-10-19:
no atmosphere/kips

Psym3trics, 2019-10-19:
im having problems with the kip ldnmitm and kip select, it worked fine befour i updated ofw from 8,0 to 9.1. i dont understand where the problem come from, i even made a fresh atmosphere install an same result.
anyone had ldn_mitm.kip working on atm 9.4?

hmax42, 2019-10-22:
can i do something to find the problem?

stuie1, 2019-10-22:
Delete whatever u downloaded what’s the problem maybe I can help

Sent from Stuart Bailey Jr

hmax42, 2019-10-25:
should i rerun Lockpick to dump 9.0.0 and 9.0.1 keys?

shchmue, 2019-10-25:
Atmosphere doesn’t care whether you’ve dumped keys or not. those are for computer programs that handle switch formats.

hmax42, 2019-10-25:
yep, did not change anything

hmax42, 2019-10-25:
so, witcher runs. the other games work again, but it doesn't make sense.

i put atmosphere *again* fresh on the msd.

SciresM, 2019-11-21:
Looks like this was solved...? Closing based on "I put atmosphere again fresh" + "other games work again".

closed 2019-11-21 by SciresM