Atmosphere-NX/Atmosphere #686: Suggestion:  Allow folders in the titles directory to have extra characters in their names after the title ID

Nazosan, 2019-10-22:
## Feature Request

### What feature are you suggesting?
#### Overview:
Allow folders in the titles directory to have extra characters in their names after the title ID itself.

#### Smaller Details: 
Currently the folder names for LayeredFS implementations must be just the title ID.  Back when LayeredFS was first implemented this wasn't even used very much.  Now not only are there many more games (and Atmosphere itself coming with several title folders of its own even) but even other things that make using the titles folders more common like cheats.  Just extracting a single cheat pack alone can flood the titles folder with too many to easily search through since cheats also use titles folders.

#### Nature of Request:  
While it would be nicest if somehow it could be text and THEN the title ID at the end, thus allowing sorting by name, I assume this would be very difficult (probably completely unrealistic) to implement.  However, it did occur to me that the other way around could be relatively easy probably.  To put it simply, just ignore everything past the sixteen digits that a title ID would use.  For example, Skyrim would normally have a folder named "01000A10041EA000" but if we could have one named "01000A10041EA000 - Skyrim" it would make it significantly easier to find it again in the future.  (And this one is a good example because, since many PC mods can be made to work with it, one may want to come back to it fairly often to make adjustments.)  Checkpoint currently does something somewhat like this, though it probably uses pregenerated names.

### What component do you feel this would best fit within?
I'm actually not sure which component this falls under.  System modules maybe?

### Why would this feature be useful?
While it still wouldn't well organize the titles, it would make finding things within them significantly faster and easier for virtually everyone.  Especially if one accesses one or more of them frequently.

SciresM, 2019-10-22:
This isn't really feasible, because title paths aren't fixed. Maybe sometime a long while in the future -- but I really want to avoid expensive path lookup at runtime via iteration over the titles dir. What happens if two folders share a title id?

mixtoism, 2019-10-22:
maybe creating short of a db relating the title id with the desired text, like a csv or something?

hippydave, 2019-10-22:
Rather than rename the folder, I just put an empty file inside it with the name of the game/sysmodule.