Atmosphere-NX/Atmosphere #689: Atmosphere fatals on first boot after factory reset

fennectech, 2019-10-27:
## Bug Report

### What's the issue you encountered?
Atmosphere fails to start after a factory reset until you complete setup in stock

### How can the issue be reproduced?
Simply factory reset the console and try to start atmo

### Crash Report

### System Firmware Version

9.0.1</br>
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Hekate
- Official release or unofficial build: 
Unoficial build
Kosmos 14.2
- Do you have additional kips or sysmodules you're loading:
SysFTPD
sysclk
bootsoundNX
### Additional context?
It all works find after completing the first setup.

fennectech, 2019-10-27:
```Atmosphère Fatal Report (v1.0):
Result:                          0x75B (2347-0003)

Title ID:                        057736ae24180000
Process Name:                    Application
Firmware:                        9.0.1 (Atmosphère 0.9.4-master-c62c4846)
General Purpose Registers:
         FP:                     0000000064c23f50
         LR:                     0000000075c01f74
         SP:                     0000000064c23f50
         PC:                     0000000075c08e68
Start Address:                   0000000075c00000
Stack Trace:
        ReturnAddress[00]:       0000000075c01ff0
        ReturnAddress[01]:       0000000075c00d7c
        ReturnAddress[02]:       0000000c8f8ef0d0
        ReturnAddress[03]:       0000000075c24ff0```
 This is the crash log

SciresM, 2019-10-27:
That is a homebrew title id. Remove your custom sysmodule, which is crashing. This isn't an atmosphere problem.

closed 2019-10-27 by SciresM

fennectech, 2019-10-28:
wait a moment
i dont think this is the right crash log

fennectech, 2019-10-28:
After retrying with plane atmosphere it still doesnt happen again so leave closed :) sorry for wasting your time.   The log was the wrong one but i cant recreate on regular atmosphere  prolly a sysmodule i have

fennectech, 2019-10-28:
Yep   OOM   why didnt i use the errbot    Sorry about that