Atmosphere-NX/Atmosphere #694: Can we disable logging?

mrdude2478, 2019-11-05:
The FAT filesystem was designed with alternating file allocation tables, so that if data is being written to the card and an error occurs (such as the switch crashing/program crashing) the alternate file allocation table is used and micro sd card corruption is minimal.

With EXFAT - this does not have an alternating file allocation table, so if the switch crashes - atmosphere writes a log to the micro sd card, and the card data can suffer from corruption as the files can't be read from the corrupt part of the file allocation table as there is no alternative to read from.

Is it possible to disable logging and writing to sd card, if the switch crashes? I couldn't see that in your readme.

fincs, 2019-11-05:
The real fix is either to pressure Nintendo to replace the sub-par exFAT driver they're presently using (prfile2), or to take it into our own hands and somehow replace it ourselves.

SciresM, 2019-11-05:
"atmosphere writes a log to the micro sd card"

It does? News to me..:

What log do you mean?

ZachyCatGames, 2019-11-05:
> "atmosphere writes a log to the micro sd card"
> 
> It does? News to me...
> 
> What log do you mean?

Perhaps they’re referring to crash/fatal reports?

mrdude2478, 2019-11-05:
> > "atmosphere writes a log to the micro sd card"
> > It does? News to me...
> > What log do you mean?
> 
> Perhaps they’re referring to crash/fatal reports?

Yes, that's what I'm referring to, if something is in the process of writing to exfat and the system crashes - file allocation table becomes corrupt. I want to disable writing of crash reports (or writing anything all all during a crash).

SciresM, 2019-11-21:
I don't think I want an ability to disable crash reports in master. 

It opens the possibility of redistributed builds not generating crash reports, which makes debugging issues hard if users use those builds. You can always edit your local atmosphere source, or just not use exFAT?

closed 2019-11-21 by SciresM

mrdude2478, 2019-12-06:
But if you can enable + disable, why would that be an issue? I'll mod the code myself then and add that function. Thanks anyway.