Atmosphere-NX/Atmosphere #695: Error when trying to install an app from the homebrew store

XianLen, 2019-11-11:
##What's the issue you encountered?

Received Error code: 2168-0002 (0x4a8)
after trying to install any app within the homebrew store.

### How can the issue be reproduced?

Tried to install a different app and issue persisted.

N/A

### System Firmware Version

9.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: official 5.0.2
- Do you have additional kips or sysmodules you're loading: NO
- Homebrew software installed: None since it wouldn't let me

### Additional context?

- Additional info about your environment:
Running CFW through EMUMMC SanDisk 128GB exFAT SD Card

austin5623, 2019-11-11:
Open hbmenu from a title. Also format your sd as fat32, exfat is asking for a bad time.

SciresM, 2019-11-11:
"2168-0002" is a userland data abort, sounds not like an atmosphere problem.

Also, "homebrew store" is not supported or recommended.

closed 2019-11-11 by SciresM

XianLen, 2019-11-11:
> Open hbmenu from a title. Also format your sd as fat32, exfat is asking for a bad time.

What do you mean open from a title?

XianLen, 2019-11-11:
> "2168-0002" is a userland data abort, sounds not like an atmosphere problem.
> 
> Also, "homebrew store" is not supported or recommended.

Could you please elaborate? I understood the homebrew store not being supported or recommended but what does the userland abort mean? do I need to fix something in my files?

SciresM, 2019-11-11:
It means that the homebrew's code is accessing invalid memory -- it's a homebrew store problem, not an atmosphere problem.

XianLen, 2019-11-11:
> It means that the homebrew's code is accessing invalid memory -- it's a homebrew store problem, not an atmosphere problem.

Thank you, now I understand the issue. I will find the app and open in a different way via .nro I guess.

fincs, 2019-11-11:
For the record, there's no such thing as an official appstore, endorsed by any of the parties responsible for making Switch homebrew possible (I.e. devkitPro, Switchbrew, Atmosphère).